ikarus23/MifareClassicTool #150: second block onward getting ALL 0's as a Value

RajuSE, 2017-10-09:
I'm trying to scan Mifare Classic tag.
I'm getting data for First Sector **0** But second sector onward getting **ALL 0's** for all sectors.
```
private SpannableString colorDataBlock(String data, boolean hasUID) {
        SpannableString ret;
        if (hasUID) {
            // First block (UID, manuf. data).
            ret = new SpannableString(TextUtils.concat(
                    Common.colorString(data,
                            getResources().getColor(R.color.purple))));
        } else {
            if (Common.isValueBlock(data)) {
                // Value block.
                ret = Common.colorString(data,
                        getResources().getColor(R.color.yellow));
            } else {
                // Just data.
                ret = new SpannableString(data);
            }
        }
        return ret;
    }
```
That's why  `(byte)(b[0]^0xFF) == b[4]) ` condition is returning false.

```
public static boolean isValueBlock(String hexString) {
    byte[] b = Common.hexStringToByteArray(hexString);
    if (b.length == 16) {
        // Google some NXP info PDFs about MIFARE Classic to see how
        // Value Blocks are formatted.
        // For better reading (~ = invert operator):
        // if (b0=b8 and b0=~b4) and (b1=b9 and b9=~b5) ...
        // ... and (b12=b14 and b13=b15 and b12=~b13) then
        if (    (b[0] == b[8] && (byte)(b[0]^0xFF) == b[4]) &&
                (b[1] == b[9] && (byte)(b[1]^0xFF) == b[5]) &&
                (b[2] == b[10] && (byte)(b[2]^0xFF) == b[6]) &&
                (b[3] == b[11] && (byte)(b[3]^0xFF) == b[7]) &&
                (b[12] == b[14] && b[13] == b[15] &&
                (byte)(b[12]^0xFF) == b[13])) {
            return true;
        }
    }
    return false;
}
```
Does that mean My card doesn't have data?
Also tell us How Values are retrieved. 

Additionally check SO I posted: [Stackoverflow](https://stackoverflow.com/questions/46582680/ndef-gettag-returns-null-mifare-support-in-android)

ikarus23, 2017-10-11:
It looks like your [device does not support Mifare Classic](https://www.shopnfc.it/en/content/7-nfc-compatibility). Mifare Classic is supported by Android, but not all NFC enabled phones have a chip that is able to interact with Mifare Classic tags. Only chips manufactured by NXP can read/write Mifare Classic.

This is supported by the fact that there was no [Mifare Classic in the TagTechnology list](https://developer.android.com/reference/android/nfc/tech/TagTechnology.html)
`Tech [android.nfc.tech.IsoDep, android.nfc.tech.NfcA, android.nfc.tech.NdefFormatable]`. If there is no `android.nfc.tech.MifareClassic` then the phone does not support Mifare Classic or the tag is not a Mifare Classic tag.

At last, I have no clue why you can "read" the tag an get `0`s for all blocks. Maybe the tag memory is just full of `0`s? Bust most likely this is because your device has trouble reading Mifare Classic.

ikarus23, 2017-10-30:
Any news on this?

ikarus23, 2017-12-27:
Lack of feedback. Closed.

closed 2017-12-27 by ikarus23