ikarus23/MifareClassicTool #152: Lenovo P2 is working (after disabling "bcm2079x-i2c" check in code)

hb9xar, 2017-11-12:
Hi,

Using a Lenovo P2 mobile phone (Android 7), the latest release version (2.1.0) claims that this device does not support Miface Classic. However, using another NFC tool (MFC Classic), I am able to read and write Mifare Classic Tags withous issues.

So I set-up an android build environment and re-built the master branch of MifareClassicTool myself with the goal to disable the "compatibility check" in the app to check if it really doesn't work. As it turns out, MifareClassicTool *does* *work* on a Lenovo P2 (P2a42, Android 7, stock ROM build P2a42_S244_170725_ROW).

I did just disable the "bcm2079x-i2c" check:

--- a/Mifare Classic Tool/app/src/main/java/de/syss/MifareClassicTool/Common.java       
+++ b/Mifare Classic Tool/app/src/main/java/de/syss/MifareClassicTool/Common.java       
@@ -597,10 +597,11 @@ public class Common extends Application {
         // This could fail because on a lot of devices apps don't have
         // the sufficient permissions.
         File device = new File("/dev/bcm2079x-i2c");
-        if (device.exists()) {
-            mHasMifareClassicSupport = -1;
-            return false;
-        }
+//&&&
+//        if (device.exists()) {
+//            mHasMifareClassicSupport = -1;
+//            return false;
+//        }
 
         // Check if there is the NFC device "pn544".
         // The PN544 NFC chip is manufactured by NXP.

Would be nice to get supoprt for Lenovo P2  into a next release.

Best regards,
/Thomas

ikarus23, 2017-11-12:
Wow what a great finding. And really annoying too. It would be really interesting to know if there is a NFC chip from Broadcom (`/dev/bcm...`) that is capable of reading Mifare Classic tags or if Lenovo just made some mistakes in their ROM.

However, I can't just remove this test. It is important to a lot of devices. And so far the P2 is the only device that is having this false positive. At least the only device I know of.

Although it's ugly, I will try to add an exception for this device in the code. Can you tell me what `Build.MANUFACTURER` and `Build.MODEL` your device has?

hb9xar, 2017-11-12:
Suggestion: Maybe an override switch in the settings/preferences that let's the user manually re-enable NFC read/write functions might come in handy?

You are right, all information that I came across indicates that the P2 uses a Broadcom chip and thus should not be able to read Mifare Classic tags. But it does work. I have not yet a satisfying explanation of *why* this is. Might be related to Chinese Mifare tags I'm using?

As my device is not rooted, I have only limited access to /dev/ (no directory read access as it seems due to selinux via adb). However, it indicates that the device file /dev/bcm2079x-i2c does exist:
```
P2a42:/ $ stat /dev/bcm2079x-i2c
stat: '/dev/bcm2079x-i2c': Permission denied
1|P2a42:/ $ stat /dev/notexistingfile                                          
stat: '/dev/bcm2079x-i2cfoo': No such file or directory
1|P2a42:/ $ stat /dev/pn544    
stat: '/dev/pn544': No such file or directory
```

Data from /system/build.prop:
```
P2a42:/ $ cat /system/build.prop |grep -i manufa                             
ro.product.manufacturer=LENOVO
P2a42:/ $ cat /system/build.prop |grep -i model                                
ro.product.model=Lenovo P2a42
ro.product.ota.model=LenovoP2a42_ROW_N_SECURITY
```
Further:
```
P2a42:/ $ ls -al /system/lib | grep nfc
-rw-r--r--  1 root root    13844 2017-07-25 11:36 libnfc_ndef.so
P2a42:/ $ ls -al /system/lib | grep brcm                                         
P2a42:/ $ ls -al /system/lib | grep bcm                                        
```

/Thomas

ikarus23, 2017-11-12:
Thanks for the information. You are right, maybe it is best to implement an option that lets you ignore the static tests (like the test for `/dev/bcm...`). My only concern are the less experienced P2 users. MCT will still tell them that their device is not capable of reading/writing MIfare Classic tags on the first launch.

So we have:
1. Add an exception to the code: More user friendly, easy to implement, not a generic solution.

or

2. Add an option to ignore the static tests for Mifare Classic compatibility: Generic solution, needs slightly more time to implement, might confuse users.

Well, I'm not sure right now which solution is better. Regardless, I will implement one of it. ;)

Rrtk2, 2017-11-16:
Hey, 
I am also able to read mifare classic 1 and 4k with the apps 'nfc taginfo' and 'nfc tools'. Sector blocks are read out as they should. However i cannot seem to write, this could be bug or faulty card (only tried one nfc-card). 

@hb9xar are you able to write in the edited MCT apk?

hb9xar, 2017-11-16:
@Rrtk2 
I'm using Chinese Mifare Classic 1k Tags and am able to properly write to them using the MCT app with the mentioned check disabled. 
I'm also able to write to those tags using other apps, such as "MFC Classic" from the app store.

Rrtk2, 2017-11-16:
Cool! Then i will try using other nfc Mifare Classic tags, to check if its just a faulty card what i used.

ikarus23, 2017-11-16:
@Rrtk2 Do you have a Lenovo P2 as well?

Rrtk2, 2017-11-17:
Yes i do! Same as you,  P2a42, Android 7, stock ROM build P2a42_S244_170725_ROW

ikarus23, 2017-11-17:
I created a [test version of MCT](http://tests.icaria.de/MifareClassicTool-2.1.0-testing9.apk) with a workaround to ignore the check for the Broadcom device on Lenovo P2 devices. Please try it and give feedback.

Rrtk2, 2017-11-17:
hey,
Ive tried the app, still deactivates the function of 'Read Tag' and 'Write Tag', not able to make it work one way or another. 

Have you only altered the following in this file: MifareClassicTool-master\MifareClassicTool-master\Mifare Classic Tool\app\src\main\java\de\syss\MifareClassicTool\Common.java.

`     File device = new File("/dev/bcm2079x-i2c");
        if (device.exists()) {
            mHasMifareClassicSupport = -1;
            return false;
        }
`
to
`     File device = new File("/dev/bcm2079x-i2c");
        if (device.exists()) {
            mHasMifareClassicSupport = 1;
            return true;
        }
`
Or did you alter the rest as well?
`   public static boolean hasMifareClassicSupport() {
        if (mHasMifareClassicSupport != 0) {
            return mHasMifareClassicSupport == 1;
        }

        // Check for the MifareClassic class.
        // It is most likely there on all NFC enabled phones.
        // Therefore this check is not needed.
        /*
        try {
            Class.forName("android.nfc.tech.MifareClassic");
        } catch( ClassNotFoundException e ) {
            // Class not found. Devices does not support MIFARE Classic.
            return false;
        }
        */

        // Check if ther is any NFC hardware at all.
        if (NfcAdapter.getDefaultAdapter(mAppContext) == null) {
            mUseAsEditorOnly = true;
            mHasMifareClassicSupport = -1;
            return false;
        }

        // Check if there is the NFC device "bcm2079x-i2c".
        // Chips by Broadcom don't support MIFARE Classic.
        // This could fail because on a lot of devices apps don't have
        // the sufficient permissions.
        File device = new File("/dev/bcm2079x-i2c");
        if (device.exists()) {
            mHasMifareClassicSupport = 1;
            return true;
        }

        // Check if there is the NFC device "pn544".
        // The PN544 NFC chip is manufactured by NXP.
        // Chips by NXP support MIFARE Classic.
        device = new File("/dev/pn544");
        if (device.exists()) {
            mHasMifareClassicSupport = 1;
            return true;
        }

        // Check if there are NFC libs with "brcm" in their names.
        // "brcm" libs are for devices with Broadcom chips. Broadcom chips
        // don't support MIFARE Classic.
        File libsFolder = new File("/system/lib");
        File[] libs = libsFolder.listFiles();
        for (File lib : libs) {
            if (lib.isFile()
                    && lib.getName().startsWith("libnfc")
                    && lib.getName().contains("brcm")
                    // Add here other non NXP NFC libraries.
                    ) {
                mHasMifareClassicSupport = -1;
                return false;
            }
        }

        mHasMifareClassicSupport = 1;
        return true;
    }
`
to

` public static boolean hasMifareClassicSupport() {
        if (mHasMifareClassicSupport != 0) {
            return mHasMifareClassicSupport == 1;
        }

        // Check for the MifareClassic class.
        // It is most likely there on all NFC enabled phones.
        // Therefore this check is not needed.
        /*
        try {
            Class.forName("android.nfc.tech.MifareClassic");
        } catch( ClassNotFoundException e ) {
            // Class not found. Devices does not support MIFARE Classic.
            return false;
        }
        */

        // Check if ther is any NFC hardware at all.
        if (NfcAdapter.getDefaultAdapter(mAppContext) == null) {
            mUseAsEditorOnly = true;
            mHasMifareClassicSupport = -1;
            return false;
        }

        // Check if there is the NFC device "bcm2079x-i2c".
        // Chips by Broadcom don't support MIFARE Classic.
        // This could fail because on a lot of devices apps don't have
        // the sufficient permissions.
        File device = new File("/dev/bcm2079x-i2c");
        if (device.exists()) {
            mHasMifareClassicSupport = 1;
            return true;
        }

        // Check if there is the NFC device "pn544".
        // The PN544 NFC chip is manufactured by NXP.
        // Chips by NXP support MIFARE Classic.
        device = new File("/dev/pn544");
        if (device.exists()) {
            mHasMifareClassicSupport = 1;
            return true;
        }

        // Check if there are NFC libs with "brcm" in their names.
        // "brcm" libs are for devices with Broadcom chips. Broadcom chips
        // don't support MIFARE Classic.
        File libsFolder = new File("/system/lib");
        File[] libs = libsFolder.listFiles();
        for (File lib : libs) {
            if (lib.isFile()
                    && lib.getName().startsWith("libnfc")
                    && lib.getName().contains("brcm")
                    // Add here other non NXP NFC libraries.
                    ) {
                mHasMifareClassicSupport = 1;
                return true;
            }
        }

        mHasMifareClassicSupport = 1;
        return true;
    }`

Or another method:
`    /**
     * 1 if the device does support MIFARE Classic. -1 if it doesn't support
     * it. 0 if the support check was not yet performed.
     * Checking for MIFARE Classic support is really expensive. Therefore
     * remember the result here.
     */
    private static int mHasMifareClassicSupport = 0;
`
To
`    /**
     * 1 if the device does support MIFARE Classic. -1 if it doesn't support
     * it. 0 if the support check was not yet performed.
     * Checking for MIFARE Classic support is really expensive. Therefore
     * remember the result here.
     */
    private static int mHasMifareClassicSupport = 1;
`

I dont know how to build the app but this is what i would change.

ikarus23, 2017-11-17:
Thanks for the feedback.
I only changed the first check to
```
        boolean isLenovoP2 = Build.MANUFACTURER.equals("LENOVO")
                && Build.MODEL.equals("P2a42");
        File device = new File("/dev/bcm2079x-i2c");
        if (!isLenovoP2 && device.exists()) {
            mHasMifareClassicSupport = -1;
            return false;
        }
```
I did not change the `/system/lib` part since @hb9xar said he tested it and removing the first check is enough. Therefore, I think it is something wrong the `isLenovoP2` variable initialization.

ikarus23, 2017-11-17:
Found it. The model is `Lenovo P2a42` not `P2a42`. Please try this [test version](http://tests.icaria.de/MifareClassicTool-2.1.0-testing10.apk).

Rrtk2, 2017-11-17:
Just tested MifareClassicTool-2.1.0-testing10.apk.

Reading works

Writing:
  Write block works
  Write dump works
  Formatting works
  Changing/Rewriting keys works
  increment works

Edit/analyse dump file works

Edit key file works

Tools:
  Tag info works
  Value block d/ecoder works
  Diff tool works
  BCC cal works

Everything works in my opinion! Thanks for the support!

ikarus23, 2017-11-17:
Great! Thank you for testing!

ikarus23, 2017-11-17:
Fixed since a167a6919b07d781eb0e72e83ebf23f087ca4af9. Closed.

closed 2017-11-17 by ikarus23

hb9xar, 2017-11-17:
A bit late, but I can also confirm that it works.
Thank you!