ikarus23/MifareClassicTool #153: Is it  possable to Using cell phone as MifareClass Card?

SuperChandler, 2017-11-13:
HCE is possable to MifareClass Card?

ikarus23, 2017-11-15:
Short answer: No.
For more information have a look at this https://stackoverflow.com/questions/20055497/emulate-mifare-card-with-android-4-4

closed 2017-11-15 by ikarus23

rawisha, 2018-05-17:
I use card emulator pro for that, works like a charm.

https://play.google.com/store/apps/details?id=com.yuanwofei.cardemulator.pro

ArchangeGabriel, 2018-05-17:
I’ve tried it, but I get an error about invalid NFC configuration file.

ikarus23, 2018-05-18:
Well, I've too seen some UID only deployments, but in Germany most of them use more than just the UID. But if you are only interested in UID + Android then this app is your way to go.

From Google Play:
>Supported Phones (with stock ROM):
Xiaomi, Huawei, OnePlus, Sony, Samsung (S4, S5, Note3), Google Phone (Nexus and Pixel), Meizu, LG, HTC, Nubia, Letv, Moto, Lenvo and maybe more?
>
>Unsupported Phones:
Samsung S6, S6 edge, S7, S7 edge, S8, S8+ and above.
NOTE: Some unsupported phone DO work with a custom ROM such as Aurora or LineageOS.

rawisha, 2018-05-18:
Gabriel, What phone do you use? I know my friend with samsung galaxy s8 have the same issue with the nfc config ... I think another ROM might fix that.

ArchangeGabriel, 2018-05-19:
OnePlus 5T, currently OmniROM but I will try LineageOS when I’ll have time for this.