ikarus23/MifareClassicTool #154: MCT don't work with emulator 1k.

maxben14, 2017-11-23:
@ikarus23 , hi.
I'm testing the development of the mifare classic emulator. The proxmark is successful do auth and read block in my emulator, but MCT does not work at  authorization in the sector. I get the error "No valid keys found".
I do snoop between my emulator and MCT with proxmark.
1208812 |    1213516 | Rdr | 60  01  7c  6a                                                  |  ok | AUTH-A(1)      
    1214768 |    1219440 | Tag | 00  00  00  00                                                  |     |
    1223564 |    1232876 | Rdr | c1 81!  1a  f0 4c! 72!  6f 4c!                                  | !crc| DEC(129)       
    1242752 |    1247488 | Tag | 96  5c  30  2f                                                  |     |
    1374908 |    1379676 | Rdr | 50  00  57  cd

I calculate timing At: (1242752 - 1232876) / 13560000 * 10^6 microsec= 728 microsec < 1 ms.
Why android send me HALT if my timing correct according to datasheet ?

ikarus23, 2017-11-24:
Hi @maxben14!  I'm assuming you talking about the ChameleonMini doing the emulation part. But I'm afraid that I'm no help to you because I don't know how Android handles Mifare Classic on the lower layers. I'm just using the Android APIs. For authentication this is function `authenticateWithKeyA()` or `authenticateWithKeyB()` which both just use the internal `authenticate()` function. You can checkout the [source code online](http://grepcode.com/file/repository.grepcode.com/java/ext/com.google.android/android/5.1.1_r1/android/nfc/tech/MifareClassic.java#MifareClassic.authenticate%28int%2Cbyte%5B%5D%2Cboolean%29) but I'm not sure if it will get you deep enough to find out what is going on.

What key is the sector using you trying to authenticate to? And what Android device are you using?

maxben14, 2017-11-24:
@ikarus23 , i talk about emutag, i write firmware for emulation 1k. I set key a0a1a2a3a4a5 in emulator and try in MCT choose key file with this key. My android 4.2 sony xperia c2005.

ikarus23, 2017-11-24:
Oh cool! The emutag is another device at my RFID shopping list ;)

I'm wondering if the Sony Xperia is working correctly. There Xperia devices that have trouble with Mifare Classic after updating to Android 5+.  Also `1208812 | 1213516 | Rdr | 60 01 7c 6a | ok | AUTH-A(1)` does not look complete, does it? Shouldn't there be 12 bytes for an authentication command?
`[60 = Auth with A] [sector number] [last 4 bytes of UID] [6 byte key A]`

maxben14, 2017-11-24:
@ikarus23 , no, android comand `[60 = Auth with A] [sector number] [last 4 bytes of UID] [6 byte key A]`
convert in standart and proxmark see such:
60 num
nt
nr ar
at

ikarus23, 2017-11-24:
Hmm, maybe it's the Xperia device after all. Like I said, I'm not familiar with that layer of Mifare Classic. It is something you don't have to touch writing apps. ;)

Have you tried another Mifare Classic compatible phone/tablet?

maxben14, 2017-11-24:
@ikarus23, no, other phone i haven't, but my android good read all card 1k and jcop with MCT.

ikarus23, 2017-11-24:
I guess other Mifare Classics apps do not work either?

maxben14, 2017-11-24:
@ikarus23 , yes, other apps don't read emutag too, but MCT reads the emulator 1k but after 5-6 attempts.
For example: 5 attempts MCT show me "not valid keys" and on 6 attempt i show sector mifare)

maxben14, 2017-11-25:
@ikarus23 , i try my device testing on acr122 successfully read 16 sectors of 1k of emulator.

ikarus23, 2017-11-25:
In that case there are two options left:
1. Android (or the device) does something wrong or
2. your emulator does something wrong.

Did you sniff the communication between the Android device and a real Mifare Classic tag to compare it to what you sniffed between the emulator and MCT?

maxben14, 2017-11-26:
@ikarus23, i think my emulator work correct, because acr122 mfclassic tool read my emulator and proxmark read my emulator.

This read 9 block in my emulator.

proxmark3> hf mf rdbl 9 a a0a1a2a3a4a5
--block no:9, key type:A, key:a0 a1 a2 a3 a4 a5
#db# READ BLOCK FINISHED
isOk:01 data:01 02 03 04 04 08 04 00 62 63 64 65 66 67 68 69
proxmark3> hf list 14a
Recorded Activity (TraceLen = 188 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |        992 | Rdr | 52                                                              |     | WUPA
       2244 |       4612 | Tag | 04  00                                                          |     |
       7040 |       9504 | Rdr | 93  20                                                          |     | ANTICOLL       
      10692 |      16580 | Tag | 01  02  03  04  04                                              |     |
      19072 |      29600 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
      30788 |      34308 | Tag | 08  b6  dd                                                      |     |
      35968 |      40736 | Rdr | 60  09  34  e6                                                  |  ok | AUTH-A(9)      
      41924 |      46596 | Tag | 00  00  00  00                                                  |     |
      56320 |      65632 | Rdr | c1 f4!  5f  1b 25! a0! 69! 61!                                  | !crc| DEC(244)       
      75124 |      79796 | Tag | 62  92  09  6c                                                  |     |
      85760 |      90464 | Rdr |5e! 23!  73 3e!                                                  | !crc| ?
     121508 |     142372 | Tag | 89 18! 09! c3!  1a  7c  a6 f7!  f8 a5! f7!  c6  94 28! 5a! 31!  |     |
            |            |     | 4e 12!                                                          | !crc|
     155520 |     160288 | Rdr |30!  ef 54! be!   

Snif beetwen real tag and android:

	3881712 |    3882704 | Rdr | 52                                                              |     | WUPA
    3883956 |    3886324 | Tag | 04  00                                                          |     |
    3895408 |    3905936 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
    3907124 |    3910644 | Tag | 08  b6  dd                                                      |     |
    4081536 |    4086304 | Rdr | 60  08  bd  f7                                                  |  ok | AUTH-A(8)      
    4088260 |    4092932 | Tag | 66  75  38  a5                                                  |     |
    4097056 |    4106368 | Rdr | 00 3f!  38 c7!  9b 2e!  18  be                                  | !crc| ?
    4107620 |    4112292 | Tag | 53 2c!  9b  77                                                  |     |
    4484400 |    4489104 | Rdr | fa  8d  e5 52!                                                  | !crc| ?
    4490356 |    4511156 | Tag | 0a  1e 8a! c2!  36  3a  36  20 97!  3e 69! 25! aa!  a9 c2! e2!  |     |
            |            |     | 8e d6!                                                          | !crc|
    4660816 |    4665520 | Rdr |71! b5!  99  ee 

I try sniff beetwen android and my emulator, but proxmark have problems sniff only with my device, proxmark loses many commands while listening and gives an incorrect trace.

ikarus23, 2017-11-26:
Hmm, really strange...
At what stage did you record this dump? If its during the key mapping process, there will be no read command send by MCT. MCT just moves on to the next sector/key once it found a matching key.

However, this is still strange. Even if you recorded this during the key mapping process, the reading should have worked. But since you got the same behavior with other Mifare Classic apps, I'm assuming it has something to do with your device or your emulator. Maybe it's best if you borrow someone else's phone, just to confirm which part might be doing something wrong.

maxben14, 2017-11-26:
@ikarus23, this sniff from my app android which only do AUTH in 1 sector.

ikarus23, 2018-05-11:
Since this looks like an issue that is specific to Android or your emulator I will close it. But you are not alone. I have issues with reading a tag with Android that is emulated by my Proxmark3...

closed 2018-05-11 by ikarus23