ikarus23/MifareClassicTool #157: Suddenly not working...

gijoe2cool, 2017-12-19:
Hello, I have been using this for the past few days now, and suddenly it's saying my device is incompatible. Why would that happen?

ikarus23, 2017-12-27:
Hmmm, strange. What device are you using? Does it work with [this testing version of MCT](http://tests.icaria.de/MifareClassicTool-2.1.0-testing11.apk)?

gijoe2cool, 2017-12-27:
Sorry, I forgot I posted here, but I eventually fixed it by restarting my phone, but it's done it once more since then. There doesn't seem to be a set situation either where it happens.

ikarus23, 2017-12-27:
Ok, thanks for the feedback. I'm closing this issue for now, but if you find out what is causing this issue, please tell me. Maybe we can fix it.

closed 2017-12-27 by ikarus23

gijoe2cool, 2017-12-27:
I will let you know the more I play with it.