ikarus23/MifareClassicTool #158: Tag lost

rawisha, 2017-12-26:
Hello, My friend just bought a huawei mate 10 pro, and when we tried this app, It reads the tags partially. 

and when we try to write info to a tag, It says "Error: Tag lost while checking for keys with write privileges"

Is there any fix for this ? 
[image]

ikarus23, 2017-12-27:
What do you mean by "partially"? Were you expecting MCT to read all data because you have all the keys? Or was this the expected behavior, because you do not have all the keys?

Regarding the connection issues: Have you tried the auto reconnect feature from the settings dialog?

rawisha, 2017-12-27:
I have all the keys and they work on my oneplus 5. Nothing weird is going on.
I have tried the Auto reconnect feature but it doesnt seem to work.

What happens is when i read the tag, some of the keys doesnt show up and some sectors says that they are dead.

If i read the tag again it can show me the info that was read before but says dead sector or cant find the keys for them .. say key A is found but B is not .. next time i read.. Key B is shown but key A is not .. or it shows dead sector.

rawisha, 2017-12-27:
This is same Tag read 3 times.

[Tag dumps.zip](https://github.com/ikarus23/MifareClassicTool/files/1589655/Tag.dumps.zip)

Just open them with text editor

ikarus23, 2017-12-27:
This sounds very familiar. I have the same inconsistent reading with my Nexus 5X. Have a look at https://github.com/ikarus23/MifareClassicTool/issues/106 It looks like this is a Android bug and there is nothing I can do about it...  :(

rawisha, 2017-12-27:
this is weird, because my friend has a p10, and it worked fine before i think when he updated the android verison, thats when it started to happen on his too.

ikarus23, 2017-12-28:
Yeah, I agree. I think it is just a software issue (and maybe a bit of a hardware issue with the antenna of the tag and the phone).

rawisha, 2018-01-01:
Wish we could find a way to fix it, is there a way to debbug it too see whats happening behind the scene.

ikarus23, 2018-01-01:
Well yes, MCT is open source. You can install Android Studio and download the source directly from this github page. You can set breakpoints wherever you want.

ikarus23, 2018-01-02:
On a side note: what version of Android is the Mate 10 Pro using?

rawisha, 2018-01-02:
Android 8, Oreo

ikarus23, 2018-01-02:
Thanks for the feedback. Let's see what will happen after the update to Android 8.1. For my Nexus 5X Android 8.1 fixed some NFC issues.

rawisha, 2018-01-02:
Sounds lovely, We will wait till mate gets the update. hopefully that fixes it

rawisha, 2018-01-15:
This can be closed, the retry authentication is set to 5 and works good both writing and reading. Still on Android 8.0

closed 2018-01-15 by rawisha

ikarus23, 2018-01-15:
Thanks for the feedback. So the Huawei Mate 10 Pro appears to have the same issue as the Google Nexus 5X (https://github.com/ikarus23/MifareClassicTool/issues/106).

rawisha, 2018-01-15:
Seems so, thanks for the update, seems to work now