ikarus23/MifareClassicTool #161: Download/Apk

Bombarus, 2018-01-01:
How i download and tranfer this in to apk , i have android studio , and git.

ikarus23, 2018-01-02:
Well, just download the code with `git clone https://github.com/ikarus23/MifareClassicTool.git` and open the `Mifare Classic Tool` folder from Android Studio as project.

ikarus23, 2018-01-02:
Hint: If you just want the Apk of MifareClassicTool you can download it [here](http://publications.icaria.de/mct/releases/)

ikarus23, 2018-01-02:
What modificated master?

Bombarus, 2018-01-02:
https://www.youtube.com/watch?v=qumhrXobI9A

ikarus23, 2018-01-02:
This code is not available here. Also there is no point in brute forcing Mifare Classic keys. As mentioned in the Youtube comments:
>Card key may have 12 hexadecimal digits (16 options per digit) 16^12 = 281474976710656 key combinations. The app searchs by 40 keys per second. This mean 281474976710656/40 = 7036874417766.4 seconds. These are 117281240296.1067 minutes; which are 1954687338.268444 hours; so it would be 81445305.76118519 days. So to be able to break a key, you might have to wait a maximum of 223137.8 years to find it. Good luck!

And the 223137.8 years are just for one of typically 32 keys...
Have a look at the getting started page of MCT:
>Due to some weaknesses in MIFARE Classic, you can retrieve all the keys (A and B) of a tag with tools like the Proxmark3 or normal RFID-Readers and some special software (mfcuk, mfoc).

Maybe it's best you start there.

Bombarus, 2018-01-02:
Thanks , have a good day :).

closed 2018-01-02 by ikarus23