ikarus23/MifareClassicTool #162: invalid acces conditions or dead sector

rawisha, 2018-01-02:
Hello, I have a zx-08cd device that can copy tags and write them. 

I have a mifare classic card with changable UID, With the program i need to decode the card and sadly it didnt work, but i went ahead and wrote what was left on the empty card .. Now when i try to read it with the MCT it shows that from sector 0 to 11 it has no data nor accessbit or key B .. but it shows key A which is valid.

normally my card as default is FFFFFFFFFFF FF078069 FFFFFFFFFFF 

and 000000.. as data.

Is there a way to restore it to that state with MCT ?

ikarus23, 2018-01-02:
Hi! Sorry I can't quite follow what you are saying. Can you make more examples? What did you exactly do? With what program? What was the data on the original tag? What's now on the new tag? What data did you write?

rawisha, 2018-01-02:
I used this [http://www.ebay.ph/itm/9-Frequency-Copy-Encrypted-NFC-Smart-Card-Cloner-RFID-Copier-ID-IC-Reader-Writer/162559752102](url) hand held machine to copy my tag and try to decode the keys to Write to a new chinese changable uid tag. "I can upload the decoder that comes with the machine"

Somehow it erased the access bits from sector 0 to sector 11 when I wrote it to the chinese tag.

Here are the dumps.

[Downloads.zip](https://github.com/ikarus23/MifareClassicTool/files/1598933/Downloads.zip)

ikarus23, 2018-01-02:
Hmmm... Interesting device. But it is hard to tell what it does. Can it really crack Mifare Classic keys in order to clone a complete tag?

If it really erased all access bits in sector 0 to 11 I have bad news for you:
>...set the access bits to an invalid value. In that case (the access bits contain an invalid value), the whole sector is permanently rendered unusable. As a security feature MIFARE CLassic cards will block access to sectors with invalid access conditions. Once a sector is in that state it cannot be recovered.

The dumps look bad. If this is a Mifare Classic Tool compatible block 0 writable tag, it is most likely broken. However, if this is a UID changeable tag with special commands ("backdoored", UID can not be set with MCT) you might have a chance. There some Chinese tags which can be set to factory settings although their access conditions are wrong. However you need special hardware/software for that. e.g. a Proxmark3.

rawisha, 2018-01-02:
how about ACR122u ?

rawisha, 2018-01-02:
and i think it is backdorred because the hand held device changed the uid .. but MCT could not even tho i tried just writing block 0 alone .. says succesfully written, but nothing changes

ikarus23, 2018-01-02:
It might be possible with the ACR122U. However, I dont't know what software to use. Over at  https://github.com/ikarus23/MifareClassicTool/issues/143 a user reports he can always format the tag just by writing an empty .mfd file.

I only tested it with a Proxmark3. There is a [special wipe command](https://github.com/Proxmark/proxmark3/blob/master/client/cmdhfmf.c#L1985) for that. However, I've not checked what commands are send to the tag.

rawisha, 2018-01-02:
I might need that .mfd file, though i am waiting for my ACR122U to arrive, hopefully it will get here soon.

rawisha, 2018-01-07:
I was wondering, Is there a way to understand the mifareclassic data and what it really means ? like actually tampering the data and change the dates, text  etc.. and is there anyway to contact you like on skype or somewhere else than here? I would like to learn more about it mifare classic and i have checked information online but dont seem to get it.

But i assume you are busy and dont have time to explain any of it.

ikarus23, 2018-01-07:
Yeah, sure, if you have the keys with write permissions and if the access conditions allow you to change data you can do whatever you want.

You can contact me via mail. You can find the address at the bottom of the Google Play Store entry of MCT.

ikarus23, 2018-05-11:
I will close this issue since wiping a broken Mifare Classic tag is not part of MCT.
Feel free to further comment on your progress. Were you able to restore the card? If so, how did you do it?

closed 2018-05-11 by ikarus23

rawisha, 2018-05-17:
Ah yes i was able to restore it, i wrote a dump from a blank card on it with an ACR122U tool, if you want, then i can send it to you.

Shikatofuka, 2019-10-31:
> Ах, да, я смог восстановить его, я написал дамп с пустой карты с помощью инструмента ACR122U, если вы хотите, то я могу отправить его вам.

Hi! Please send me dump =)