ikarus23/MifareClassicTool #163: Mifare mini s20

Aziz1917, 2018-01-03:
Could you add support for mifare mini? Can not copy the dump. The program does not see one of the keys in the sector.

ikarus23, 2018-01-03:
MIFARE Mini should be supported. The reason MCT "does not see the keys" might be because it does not know the keys. MCT can not crack any keys. You have to know them or extract them yourself. Have a look at the [readme](https://github.com/ikarus23/MifareClassicTool#getting-started).

Aziz1917, 2018-01-03:
The program for some reason does not see key B. Although mfoc and bitsliced ​​write that it is true.

  
  

-- Отправлено из мобильного приложения Яндекс.Почты

ikarus23, 2018-01-03:
What do you mean by that? Did you recover key B with mfoc and added it to a key file of MCT? Did you try to read the tag with the recovered key and other software/hardware?

Aziz1917, 2018-01-03:
Yes, I found the B key via mfoc and added it to the MCT. However, MCT thinks that key B is wrong.

ikarus23, 2018-01-03:
What device are you using? Have you tried to read the tag with the external RFID reader you used with mfoc?

Aziz1917, 2018-01-05:
Oneplus one. It works well with other types of cards, but the problem with the B key is in mifare mini. MCT can not read it

Aziz1917, 2018-01-05:
Read with the help of external rfid reader did not try

ikarus23, 2018-01-05:
Oh, the OnePus One. This device is known for trouble with Mifare Classic! What ROM are you using? Users reported  it should work with the latest version of LinageOS.

Aziz1917, 2018-01-06:
I'm fine with all types of cards except mifare mini.

ikarus23, 2018-01-06:
Even so, there still is a possibility that the key is wrong or the Android ROM you are using has trouble with Mifare mini. Please verify that you can read the tag with another setup (e.g. ACR 122U + libnfc). Also, please verify that the access conditions of the sectors allow you to read the blocks with key B.

Cuk-o, 2018-01-29:
join to issue. same situation. using mi5 (xiaomi ) also tried OneplusOne

ikarus23, 2018-01-29:
@Cuk-o, how did you recover the keys? Are you sure as well that your keys are right? Did you try to read the tag with an USB reader? I just want to rule out any mistakes before locking at it more closely.

Cuk-o, 2018-01-29:
i get keys by "Bitsliced Crypto-1 brute-forcer". also mfoc telling me keys is ok. but MCT and nfc-mfclassic cannot read and make full dump of card. on pc i use pn532 on kali64. another keys(1k) work fine.

ikarus23, 2018-01-30:
Interestings, so `nfc-mfclassic` can not read it either. I think there might be an issue with the mini tag if other tags are working fine.

Cuk-o, 2018-01-30:
i have couple mini tag, about 6 pcs. same situation. this tag works with my elevator. what can i do?

ikarus23, 2018-01-30:
Well, a first step would be to find out what is causing the issues.
* Is it the tags? Is there something special about them? Are those original tags? Are they emulated?
* Is it the setup? Have you tried other reading software? Have you tried to make it work on Windows?
* Are the recovered keys correct? Have you any other chance to verify this? Maybe access to the official reader or the software for that reader? Maybe the keys are well known and can be found by a Google search?

But apart from that, there is not much I can do. I do not have a Mifare Classic Mini tag. If you don't need one of the tags in question, you can send it to me and I will have a look at it.

Cuk-o, 2018-01-30:
i would be glad to send it to you
yes its a tag. regular mini tag. not sure about its original or not.
i tried any software what i can find on linux and windows. also tried miLazyCracker.
about verify. A keys in MCT and mfoc tell me its right. so that way what i thinks B keys good also.

ikarus23, 2018-01-30:
Have you checked the access conditions? What permissions does key B have?
Regarding sending the tag: Just write me an e-mail and I will give you an address. You can find the mail address on the google play page of MCT. However, I can not promise any success.

Cuk-o, 2018-01-30:
anyway thx for trying for help.
i dont know how to check)

ikarus23, 2018-05-11:
Any news on this? I still don't have any Mifare mini tag...

Cuk-o, 2018-05-11:
Hello, i havnt been at home, will send it soon.

Cuk-o, 2018-09-11:
Hello friend, I will get cards after 19 of Sept. Sorry for so long time.
It's same address ?

ikarus23, 2018-09-11:
No worries. Yes it still is the same address.

Cuk-o, 2018-09-26:
Hello dear, today I get one card and send it to you. Track number is
RR016726935KZ. It's airmail by Kazpost. They told me maximum is 2weeks.
Hopefully you will get it soon.

ikarus23, 2018-10-08:
Hi, the package arrived today. Unfortunately, I got the same results so far.
* Sector 0 has default keys and ACs.
* Sectors 1-4 have the same key set.
* It is possible to read data from sector 0.
* It is possible to authenticate to sector 1-4 with the recovered key sets.
* Although the keys to sectors 1-4 are valid for authentication, reading data fails.

I tested this with a Proxmark3, a Nexus 5X and an ACR122u. All come to the same conclusion. At his point, I'm not sure it the tags are partially bricked. This is strange as there are 6 tags with this behavior. However, @Cuk-o says that his tag is still working in the elevator. This strengthens the assumption that the reader in the elevator just uses the UID. Furthermore, the UID is printed on the tag.

@Cuk-o Do you have any UID changeable tag?

Cuk-o, 2018-10-08:
Hello, yes I have

ikarus23, 2018-10-08:
Could you try to change the UID  to the UID of one valid working tag? Nothing more (no data editing/copying). Maybe it will work in your elevator.

Cuk-o, 2018-10-08:
Ok I will try, and reply after

Cuk-o, 2018-10-08:
Did you try to get all keys by *Bitsliced Crypto-1 brute-forcer? *

Cuk-o, 2018-10-08:
I changed UID for new key, unfortunately  it's works...

ikarus23, 2018-10-08:
>unfortunately it's works...

I don't get it. Did it work or did it not work?

Yes I got the keys for the tag. But like you I'm not able to read it with them.

Cuk-o, 2018-10-08:
It works with just UID, I wanted to change number of rides, but now I get,
rides keeps on server.

ikarus23, 2018-10-09:
Great! And yes, it is most common for "UID-only" systems to have a backend database with all the data.

ikarus23, 2018-10-09:
You are welcome. I will close this issue now although there the unanswered questions of "why are there sectors with working authentication but broken reading". This question is not related to MCT. I think the Proxmark3 Forum might be a better place to discuss this.

closed 2018-10-09 by ikarus23