ikarus23/MifareClassicTool #166: CRC check (cyclic redundancy check)

ai909554, 2018-01-06:
Do you have the idea to choose to add a CRC check？Think of the software BBC calculator（I am not very good at English
）

ikarus23, 2018-01-06:
As much as I like the idea, I don't think I will implement a good and easy to use CRC tool in MCT. I just don't have the time for it. Also, there are really good external tools for this. Check out [reveng crc](http://reveng.sourceforge.net/).

ai909554, 2018-01-06:
I really can't stand using an online CRC calculator on a mobile device.I was looking for a convenient and quick CRC calculator, but I didn't find it.Can you tell me if I have any software like this？

ikarus23, 2018-01-06:
I understand, but unfortunately I don't know any good CRC calculator for Android. There is [Hashr](https://play.google.com/store/apps/details?id=com.misker.mike.hasher) which can calculate CRC32 of files and text. But I don't think this will help you much.

You could also try to "download" an Javascript-based online CRC calculator with tools like [HTTrack](https://play.google.com/store/apps/details?id=com.httrack.android). I never tried it though.

I will keep this issue open as feature request for a CRC calculator tool. But I can't say if and when I will have the time to write such a calculator and  what CRC algorithm it will support.

target111, 2018-06-21:
Much appreciated if you could include this

sesam, 2018-09-04:
Which CRC version is needed? Classical CRC32, or something else?

target111, 2018-09-06:
something like [this](https://www.lammertbies.nl/comm/info/crc-calculation.html) would be perfect