ikarus23/MifareClassicTool #169: Diff dump not working on new releases

lambia, 2018-01-21:
Hi there, I just updated to the new release, from an old one of months ago (maybe 2.something?).
I noticed that the hex editor doesn't work properly anymore when comparing two dumps.

If you open a dump and then you choose to compare it with another, just won't do it, asking you to choose a file in a endless loop.
If you go to tools and then to compare option, you can't fill both the file selection. Choosing A will delete B and so on.

Sorry for my bad english, anyway if this problem is not happening just on my phone, you should see it and understand of what I am talking in seconds.

ikarus23, 2018-01-23:
Hi @lambia, thanks for reporting. However, on my Nexus 5X and on my Nexus 7 (2012) this issue does not occur. What device are you using? Also, you can install any old version from https://publications.icaria.de/mct/releases/ and test what exact version break it.

ikarus23, 2018-02-21:
Any news on this?

lambia, 2018-02-24:
Sorry for the delay. Version is 2.2.3 on LG G4 with Nougat 7.0.
I tried reinstalling, without success. I tried the last 5 releases and the problem is still there.
But I start believing it's my phone fault. I'll try to reset this phone rom asap (next month, surely).
The same version isn't affected on Nokia 8, Galaxy S8 and ZTE Axon 7 (all with Nougat 7.1).

ikarus23, 2018-02-25:
Thanks for replying! And thanks for the extended testing on Nokia 8, Galaxy S8 and ZTE Axon 7. If you are able to find out what exactly is causing this behavior, feel free to re-open this issue. Maybe it is MCT's fault after all and maybe we can do something about it. ;)

closed 2018-02-25 by ikarus23