ikarus23/MifareClassicTool #173: How read NTag in This example?

Jiks19, 2018-02-24:
Hello,

I tried NFC ntag type using this example.
But can't read it.

Please can you give me any suggestion/help?

Thanks.

ikarus23, 2018-02-25:
Well, this app is named **MIFARE Classic** Tool. As you can read in the README:
>This tool provides several features to interact with (**and only with**) MIFARE Classic RFID-Tags.

In other words: This app can not read NTags. It's only for MIFARE Classic.

closed 2018-02-25 by ikarus23

Jiks19, 2018-02-27:
Thanks for replay.

I have ACR122U external USB device.
I tried read Ntag with external USB device.

ACR122U device mifare cards takes more time to read.
So we need to read ultralight cards.

please give me suggestions / help.

ikarus23, 2018-03-01:
Sorry, I do not understand your question. What are you trying to do? What's with the read time?