ikarus23/MifareClassicTool #175: Testing new hardware.

levoniust, 2018-03-08:
I have the Samsung Galaxy Note 8 from AT&T SM-N950U. The app on Google Play says my phone is not compatible, but when I went through this for him I did not see my phone listed for functioning or non-functioning. Is there a location I can go to see if my phone has the correct Hardware or software so I can run this application? Of course I downloaded the APK and it appears to read correctly but I don't know if it can write.

ikarus23, 2018-03-10:
Hi,

as far as I know, the Note 8 is not supported. It is on the [list of incompatible devices](https://github.com/ikarus23/MifareClassicTool/blob/master/INCOMPATIBLE_DEVICES.md) because users reported to me, that it didn't work.

What do you mean by "it appears to read correctly"? Can you read a full tag? Or just the UID and some other generic stuff? I heard of some reports saying the Galaxy S8 started to work with Mifare Classic after the latest Android update. Maybe it's the same for the Galaxy Note 8.

levoniust, 2018-03-10:
I have no idea. I saw that it read something, I wish I knew more of what I was looking at. But all I was trying to do was read and emulate my hotel key. Tell you want if you send me a amazon link to some nfc cards that use the midfare classic standard ill buy them and report back to you.

arnemauer, 2018-03-11:
Since oreo update Samsung s8 supports reading id and data of mifare classic cards.

gmkbenjamin, 2018-04-27:
@arnemauer 
Same here. I have Samsung Galaxy Note 8 SM-N950F variant in Australia, all apps started working with Mifare Classic after upgrading to Oreo, this includes NFC taginfo and MCT.  Please remove

ikarus23, 2018-05-11:
Let us continue the discussion about working or not working Galaxy Note 8 devices at  https://github.com/ikarus23/MifareClassicTool/issues/181. Closed.

closed 2018-05-11 by ikarus23