ikarus23/MifareClassicTool #179: Error when enable writing to manufacturer block.

syshut, 2018-04-11:
Hi, thank you for this great tool! But have ran into a problem when using it.

When I enabled the option "enable writing to manufacturer block", there would appear a message saying "Error: Some error occurred while writing. This could be really bad." And the dump is not written into the card.

But if I canceled this option, the dump can be written correctly. 

I am a primary user of MCT, and I am not very good at speaking English. I hope I have describe the problem clearly and you can understand what I am saying. If so, would you please tell me how to solve it?

ArchangeGabriel, 2018-04-11:
What kind of tag are you writing to? Only some specific chinese magic tag can have their manufacturer block rewritten.

syshut, 2018-04-12:
I am writing to an Chinese magic card, to which I can write the manufacturer block with my PM3 device.

ArchangeGabriel, 2018-04-12:
They are two kinds of Chinese magic card, and MCT can only write manufacturer block on one of the two. I don’t know how to differentiate them technically, but @ikarus23 could likely tell you how to determine the kind of magic card you have.

RasmusKallas, 2018-04-12:
I bought [10x cards off ebay](https://www.ebay.co.uk/itm/10pcs-13-5MHZ-UID-Changeable-M1-S50-1K-NFC-Card-Copy-Rewritable-Blank-IC-Card-BB/202173611232?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649), but they don't work either, interested where you could get rewritable uid cards, what work with MCT (since I won't buy a pm3 for 3 cards).

ArchangeGabriel, 2018-04-12:
@RasmusKallas They are cheaper option than a pm3 you know… I don’t know if you can have the required information from sellers (since to begin with I don’t know what is the required information, just that they are two kind of magics tags and that some of mine works and some don’t).

RasmusKallas, 2018-04-12:
@ikarus23 Thanks, it found results for 'MCT' 👍

syshut, 2018-04-13:
@ikarus23 
Do you mean <code>hf 14a reader</code>?
<pre>
proxmark3> hf 14a reader
ATQA : 00 04          
 UID : 72 ** ** 01           
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1          
proprietary non iso14443-4 card found, RATS not supported          
Answers to chinese magic backdoor commands: YES          
</pre>

syshut, 2018-04-13:
But why I can write the 0 block with pm3? Can MCT also do this?

rawisha, 2018-04-13:
That is backdoored one, you need special commands for it which android can't do.

MCT only works with direct write.. 

Correct me if i am wrong.

ArchangeGabriel, 2018-04-13:
@syshut @rawisha is right, this is a limitation of Android, not MCT.

rawisha, 2018-04-13:
Been ordering several cards lately but none if them seems to be genuine, anyone knows where i can get a card or two with direct wirte?

ArchangeGabriel, 2018-04-13:
@rawisha Just above @ikarus23 advised to look for “mifare MCT” in offers.

rawisha, 2018-04-13:
Indeed, just didn't find any

RasmusKallas, 2018-04-13:
@rawisha AliExpress: search for 'MCT' you will find results with *** android app mct

RasmusKallas, 2018-04-13:
(mifare mct has no results, but mct does)

ikarus23, 2018-04-13:
@syshut: `hf 14a reader` is ok too. `hf 14a info` is available in new firmware versions and displays more information (e.g. if the PRNG of the tag is weak). However, like @rawisha and @ArchangeGabriel said, it is not possible to change the UID of these tags with Android. You need a Proxmark3 or an USB reader like the ACR122u.

Regarding the tags: Just like @RasmusKallas  said, buy tags from AliExpress. Search them by using keywords like "MCT" or "Android". I also bought some on ebay and they worked too. For example [these might work](https://www.ebay.com/itm/10PCS-13-56MHz-IC-Writable-Card-UID-Changeable-1K-Card-block-0-MCT/323196861697?hash=item4b4008bd01:g:GaAAAOSw8YZZ1JWK). But no guaranties and there are even cheaper ones from time to time.

closed 2018-04-16 by syshut