ikarus23/MifareClassicTool #180: How come I can read whole sector but key is empty?

positron96, 2018-04-15:
Hello. 
I have a seemingly blank mifare classic card that I am trying to read.
I have read the whole card except the key for sector 0. The default ACs say that only keyA can be used to read the sector (data blocks as well as ACs and keyB are read), so it obviously was used to read the sector (and most probably it is FFFFFF as the card seems to be blank). However the app shows that it has no keyA information. Why is this?

What's more interesting, in other sectors keyA field is populated correctly. Why are keyA fields in sector 0 and others different? 

Is it a bug or am I missing something? 

My device is Sony Xperia Z3 Compact (Android 6.0.1).

[image]

ikarus23, 2018-04-20:
Hi there!
You are right, it really looks like a bug. Even if the missing key is not FFFF... it should be there, because it was used for reading. Another thing I havn't seen so far is the second block being identical to the first block.

Could you please try another app that is able to read Mifare Classic tags with default keys? For example [NFC TagInfo](https://play.google.com/store/apps/details?id=at.mroland.android.apps.nfctaginfo) or [NFC TagInfo by NXP](https://play.google.com/store/apps/details?id=com.nxp.taginfolite). It would be interesting to see if they are able to read the tag without any issues.

positron96, 2018-04-21:
Hello and thank you for response!

I've tried NXP TagInfo and got exactly same result. So if it looks like a bug, it's probably device- and firmware-related. 
If you are willing to make a workaround in your app, I can provide some assistance on my device.

[image]

ikarus23, 2018-05-11:
Do you have access to a different Android device that supports MCT? I would be interesting to see if the tag in question is read the same way on other devices.

ikarus23, 2018-09-10:
Any news on this?

positron96, 2018-09-11:
Unfortunately, no new Android devices around me.

ikarus23, 2018-12-28:
I will close this issue for now. It looks like a device bug and not a MCT bug. Feel free to re-open it if you have news.

closed 2018-12-28 by ikarus23

positron96, 2019-07-09:
Hello. I've finally managed to test my tag on another device and it gives same result with key hidden but the whole sector read successfully. I've tested on Highscreen Power Five Max 2. Scanned map is attached.

[UID_968CB79E_2019-07-09_14-04-18.txt](https://github.com/ikarus23/MifareClassicTool/files/3372640/UID_968CB79E_2019-07-09_14-04-18.txt)

ikarus23, 2019-07-09:
Interesting. Maybe it is the tag which is not behaving according to specification...

positron96, 2019-07-10:
Must be.