ikarus23/MifareClassicTool #181: Won't Read/Write After Android Oreo Update

amgalitzky, 2018-04-19:
I use a Samsung Galaxy Note 8 i had Android 7 & everything worked great.

Now i updated to Android 8 (Oreo) & ever since I can't download the app from the play store. I'm getting a message "Your device isn't compatible with this version".

After downloading it from github it wouldn't read/write tags it just keep on loading as if it's reading/writing but never finishes.

amgalitzky, 2018-04-20:
This is indeed strange. I was using it with Android Nougat without any issues, everything worked. And I was able to download it on Google Play too. ever since the update i have this issues.

I just tried testing NFC Taginfo & NFC Taginfo by NXP and i have the same issue there, the phone is detecting the NFC tag and can get the basic tag info but it works read/write/format.

amgalitzky, 2018-04-22:
Thanks let's hope they will fix it in a future update.

norbital, 2018-04-25:
Same with me.  Copied and wrote cards no problem... All nfc apps except pay apps are broken after Oreo... Thanks Samsung

norbital, 2018-04-25:
Oh... Just wanted to thank you for your work. Great app while it worked for me

amgalitzky, 2018-04-25:
I contacted Samsung and after some testing on their behalf they acknowledged in the issue and they told me they apply a fix in a future update.

ikarus23, 2018-04-25:
@norbital You are welcome ;)
@amgalitzky Thanks for reaching out to Samsung! I will remove the Note 8 from the blacklist as soon as the patch is rolled out.

gmkbenjamin, 2018-04-27:
I have Samsung Galaxy Note 8 SM-N950F variant in Australia, all apps started working with Mifare Classic after upgrading to Oreo 8.0.0 build number: R16NW.N950FXXU3CRC7 (without any additional patches), this includes NFC taginfo and MCT.  Please remove it from the blacklist.

ikarus23, 2018-04-27:
What now? @amgalitzky says the Note 8 does not work after updating to Oreo any @gmkbenjamin is saying it started working on the Note 8 after updating to Oreo?

Also, there are people saying it started to work on S8 and some other say it does not. See https://github.com/ikarus23/MifareClassicTool/pull/172.

Are the that many version of the Note 8 / S8 with different Hardware/Software?

amgalitzky, 2018-04-27:
I have a different build number my built number is R16NW.N950U1UEU3CRC1 i guess in Australia they have different software's

gmkbenjamin, 2018-04-27:
Yep, there are 2 chipset versions and multiple carrier specific software version. I was so disappointed that I couldn't use this tool when I switched from pixel. I'm so happy now that I can use it again ;)

gmkbenjamin, 2018-05-11:
Mine is greatlte. And it works on Android 8. I'm not sure if carrier specific ROM comes into play though......

norbital, 2018-05-11:
It seems to. Mine is greatqltecs (Canadian) and it generally doesn't work... 
If I mess with it turning off and on NFC then immediately reading or writing a tag I can get it to work, but only a 1 in 10 success rate.

ikarus23, 2018-05-11:
Thanks for the feedback @gmkbenjamin and @norbital!

I will wait a little longer before unlocking `greatlte`. Maybe someone with another ROM but the same model has more feedback. 

@norbital Well, this sounds like a solid bug. And an annoying one too! Lets hope it gets fixed due to @amgalitzky's feedback to Samsung.

amgalitzky, 2018-05-15:
I just got a software update (Version: N950U1UEU4CRD7/N950U1OYM4CRD7/N950U1UEU4CRD7) which did still not fix the issue.

eslavnov, 2018-06-19:
@ikarus23 Thank you for a great app!

I have a Dutch Note 8 (greatlte) with original Oreo ROM (Build number: R16NW.N950FXXU3CRE5) and MCT version 2.2.3 works fine for me. Would be awesome if you could unblock it on Play store!

taylorwin, 2019-03-01:
> I added the Samsung Galaxy S7, S8 and Note8 to the list of compatible devices and made MCT available to them in the Play Store.
> 
> I hope the poor users of some bugged S8 and Note8 devices will receive their update soon so that they can use MCT as most other S8/Note8 users.
> 
> I will leave this issue open for discussion and feedback on the rollout of the update with the fixes.

Hi ... I use note8（R16NW.N9500ZCS4CRL1）.
Upgraded to the latest through the Google Store, but still failed to read.
However, the data can be read normally using https://why.yuyeye.cc/mtools/en/index.html.
Hope it helps you

taylorwin, 2019-03-13:
Upgraded to the latest through the Google Store, but still failed to read all block one time.

help me …………

ikarus23, 2019-03-16:
Have you tried the "retry authentication" option?

taylorwin, 2019-03-19:
> Have you tried the "retry authentication" option?

thank you for your reply

 "retry authentication"  Set it the value is 1 .but the same failed.
The status is  Hang up.

ikarus23, 2019-03-24:
I'm sorry, but this really looks like an issue of your device or the ROM, but not MCT. I don't think there is something I can do.

taylorwin, 2019-04-03:
I did some testing。
1. Just use keyA  , MCT crashed.
2. Just use kenB , MCT crashed.
3. Use together keyA,kenB it work ok.

But i use Another mobile phone , just user keyA  it is work ok .
All test use the same card.

ikarus23, 2019-04-24:
Very strange. It might be connected to the other issues with only one key in the key file: https://github.com/ikarus23/MifareClassicTool/issues/66.

ikarus23, 2020-08-16:
I will close this now. It seams MCT works now on most of the Note 8/S7/S8 devices. Even if not, there is nothing I can do. It looks like there still some issue with some ROMs for specific models or countries.

@taylorwin Your issue could be different. If nothing has changed since April 2019 and even the latest [testing version](http://tests.icaria.de/MifareClassicTool-2.3.1-testing4.apk) does not work, you could open another issue.

closed 2020-08-16 by ikarus23