ikarus23/MifareClassicTool #185: Sector 0 - Corruption

fabltd, 2018-05-16:
After editing sector 0 and writing the dump.

Card now reports: No Keys Found (or dead sector)

Is this reversable?

rawisha, 2018-05-17:
I have an ACR122U and there is a tool that follows, does the job. Just use a dump from a working card and replace it.

ikarus23, 2018-05-21:
Yes, try a reader like the ACR122U. Also, what exact 4 blocks did you write into the first sector?

uchihamadara23, 2018-07-02:
I have used the app to read my card but when it finish only the sector 0 is visible... it have 5 sectores (0 to 4) but the sector 1, 2, 3 and 4 says "No keys found (or dead sector)"... what that means?

ikarus23, 2018-07-03:
@uchihamadara23, well it means exactly that. None of the keys in the key file you provieded contained a valid key. You can't just read a Mifare Classic tag. You need the keys (passwords). Have a look at the [Getting Started](https://github.com/ikarus23/MifareClassicTool#getting-started) section of the readme file.

And there is also the "dead sector". If you wrote invalid access conditions to a tag, the sector will permanently break and will no longer be usable.

ikarus23, 2018-09-10:
As this is most likely not related to MCT, I will close it. Feel free to document your success/error in trying to reverse the changes you've done.

closed 2018-09-10 by ikarus23