ikarus23/MifareClassicTool #189: No keys found or dead sector

uchihamadara23, 2018-07-02:
I have used the app to read my card but when it finish only the sector 0 is visible... it have 5 sectores (0 to 4) but the sector 1, 2, 3 and 4 says "No keys found (or dead sector)"... what that means?

ikarus23, 2018-07-03:
Well it means exactly that. None of the keys in the key file you provieded contained a valid key. You can't just read a Mifare Classic tag. You need the keys (passwords). Have a look at the [Getting Started](https://github.com/ikarus23/MifareClassicTool#getting-started) section of the readme file.

And there is also the "dead sector". If you wrote invalid access conditions to a tag, the sector will permanently break and will no longer be usable.

uchihamadara23, 2018-07-03:
So in this case, to read all information in tag I need the Proxmark3 tool or ACR122U isn´t it? If I have the ARC122U I will be able to read all information on the tag? and possible clone or edit information?
thanks a lot friend :)

ikarus23, 2018-07-05:
Yes, you should be able to crack the key with the ACR122U and mfcuk/mfoc. At least if the tag is not the new generation with enhanced PRNG. Just try it. If you got the keys you can edit the data. You can also copy the tag to an empty tag. If you want to do a perfect clone you need a magic chinese card because they allow you to change to UID/edit block 0.

uchihamadara23, 2018-07-10:
Friend, can u help me please? :)

ikarus23, 2018-07-10:
I don't know if it will work with MIZIP Plus. I never head access to such a system/key. Sorry I can't help you with that.

TheVampkid, 2018-09-03:
I've come across this as well. I understand it's reversible with proxmark3 but is it reversible with mfcuk

ikarus23, 2018-09-05:
What issue do you have? Don't you have all the keys? Do you use a MIZIP Plus? Well, you need the keys to read a Mifare Classic tag. And yes, you can get the keys with proxmark3 or mfcuk.

ikarus23, 2018-09-10:
Closed as off-topic (not related to MCT).
However, if you have a concrete question I will try to help you as good as I can.

closed 2018-09-10 by ikarus23