ikarus23/MifareClassicTool #190: No keys found (or dead sector) android 8.0 + MI6

Aqho, 2018-07-10:
Hello, previously i have a one plus one, he just finish he's life :'( RIP him,
So i'm now on a Xiaomi MI6 (global 6Go version) and MCT seems to have some trouble with it.
I'm actually on Miui 10 Global android 8.0.0 and i have this error(same trouble on miui 9):
using a mifare classic 1K and only on sector 14 - 15
No keys found (or dead sector) 

the keys are currently 000000000000 and FFFFFFFFFFFF on these sectors and mct can't read it, other app like MFClassic arrived with the same keys...
[image]
[image]
[image]

thank's for reading me :)
Nice work on mct btw, awesome tool!

ikarus23, 2018-09-10:
Any news on this? Could you try apps like "NFC TagInfo"?

ikarus23, 2018-10-02:
Lack of feedback. Closed.
Feel free to give feedback/reopen the issue once you tried other apps like "NFC TagInfo"!

closed 2018-10-02 by ikarus23