ikarus23/MifareClassicTool #192: [Question]Read wiegand data (Facility code and card number) from Mifare cards?

gmkbenjamin, 2018-07-11:
Hello,
I could be very wrong here. When Mifare Classic/DESFire cards used in physical control systems e.g. doors, I'm assuming the system still uses wiegand data to authenticate the card? If so is it possible to read this data from a smartphone?

Thank you.

ikarus23, 2018-07-19:
Sorry, I don't know. I never stumbled across a Mifare Classic/DESFire system with wiegand. They are not this popular in Europe (anymore?).

gmkbenjamin, 2018-07-20:
The desfire card we use for our office uses Wiegand(facility code and card ID). I tried many tools and readers I just can't figure out how to read the facility code and card ID. But I can read it on our physical access control system Gallagher. 

Do you mind if leave this issue open for a bit in case someone here knows? Thank you.

ikarus23, 2018-07-29:
>I tried many tools and readers I just can't figure out how to read the facility code and card ID.

Well, DESFire cards are organised in applications and files. They can be protected with different (AES/DES) keys. Therefore it is most likely you can't just read the data, you need the keys first. You can use a NFC enabled phone and try [NFC Taginfo](https://play.google.com/store/apps/details?id=at.mroland.android.apps.nfctaginfo) to get some information of the card.

(I will leave this issue open for some time.)

ArchangeGabriel, 2018-07-30:
@gmkbenjamin Generally, yes, most likely at some software level (most of the time readers granting facility access are network-linked to some controller).

ArchangeGabriel, 2018-07-30:
When used properly, DESFire cards are well-secured (I think at the most there was some side-channel attack on the earliest version, but I think that this has been patched a while ago by the release of EV2).

ikarus23, 2018-07-30:
As @ArchangeGabriel said, the keys are most likely configured in some backend software. And yes, if DESFire is done right, there is no hack that I know of to crack EV1 and EV2. (The side-channel attack only worked for DESFire prior to EV1).

samscode, 2018-08-26:
I think perhaps the confusion stems from Wiegand format vs. Wiegand protocol. Gallaghar contactless readers can still be configured to read Wiegand format cards, or cards with data encoded in all sorts of weird and wonderful ways actually.

Gallaghar use the terminology "faculty code" and "card id" on modern, secure smart card solutions (DESFire). If you check the Wikipedia page about Wiegand, you'll probably find that your organisation couldn't possibly be using the "8 bits of facility code, 16 bits of ID code" as mentioned there.

ikarus23, 2018-09-10:
Closing this issue. However, feel free to exchange more information on DESFire and Wiegand ;)

closed 2018-09-10 by ikarus23