ikarus23/MifareClassicTool #197: Feature request: support for app "External NFC Reader Service" by Thomas Rorvik Skjolberg

sesam, 2018-09-04:
This app https://play.google.com/store/apps/details?id=com.skjolberg.nfc.external
supports more chipsets than the already supported app https://play.google.com/store/apps/details?id=eu.dedb.nfc.service

Skolberg's app claims it is easy to use. Check out the public API at https://github.com/skjolber/external-nfc-api

ikarus23, 2018-09-04:
Sounds interesting. I will have a closer look into this. But first I have to attend some other issues (Android 9 is making trouble). Also:
> For ACR 122U the Mifare Classic does not work well.

The ACR 122U is exactly the reader I own...

ikarus23, 2019-01-31:
There has been an update to External NFC and I made some adjustments in MCT. Here is a [testing version](http://tests.icaria.de/MifareClassicTool-2.2.4-testing6.apk) which should work. I was able to read and write a Mifare Classic tag with my ACR122u on Android.

I think I will let this one slide in favor of "External NFC". It's available in the Play Store for free and I don't have to implement something to make it work.

If someone is willing to implement the support for "External NFC Reader Service", please do so and issue a pull request. ;)

closed 2019-01-31 by ikarus23