ikarus23/MifareClassicTool #198: Request: documentation how to use external NFC reader (writer?) via USB-OTG capable phones

sesam, 2018-09-04:
As I understand, MifareClassicTool already supports https://play.google.com/store/apps/details?id=eu.dedb.nfc.service which in turn supports:
 - MFRC522 via USB-UART adapters.
    - Probably $1.50 for the MFRC522 + $5 for a USB-OTG compatible UART?
 - ACR122
    - around $25 from China/HongKong. But there seems to be several versions of ACR122, some even with bluetooth. Which ACR122 models have been tested and working?

ikarus23, 2018-09-04:
Hi! Unfortunately I have trouble with both. The MFRC522 setup and the ACR122. The issues are not the same. I use an ACR122U (with USB). If I remember it correctly (I can't test right now) I was able to discover tags with the ACR122U. Maybe even read them. But I definitively  head trouble with writing.

I contacted the author of "External NFC" several times in hope to improve the app. But just like me he is an hobbyist who's short on spare time...

kgamecarter, 2018-12-20:
How about direct use ACR122U sdk in MCT.

ikarus23, 2018-12-20:
I'm not sure about that. I can't find any legal information (can I use it? can I distribute it?, etc.). Also, I don't like the idea of having some unknown software blob in this open source app. Maybe implementing it like a plugin would be appropriate. But I'm not sure if I will find the time to do it.

kgamecarter, 2019-01-04:
Maybe libusb+libnfc is a solution.

ikarus23, 2019-02-01:
@kgamecarter confirmed in https://github.com/ikarus23/MifareClassicTool/issues/226 that it is working for him too.

I haven't checked the MFRC522 yet, so I will keep it issue open for a bit. But I don't think there is much documentation to be done here. If External NFC works, MCT should work with it too.

kgamecarter, 2019-02-02:
I test PN532 with UART is work fine.
[image]

ikarus23, 2019-02-02:
This is great! Thanks for testing!

closed 2019-02-16 by ikarus23

yashtriv13, 2020-01-04:
Hey, I've been trying to use the MCT app with the MFRC522 but it seems to me that the USB UART adapter i'm using is creating problems. I used one with CP2102 chip. Is there an adapter that you have already tested and works?

ikarus23, 2020-01-13:
Sorry, I have not tested a CP2102 based UART-to-USB adapter.