ikarus23/MifareClassicTool #200: [Android 9] Black text on dark grey menu

bennesp, 2018-09-24:
As title, each menu of the app on Android P (on Oneplus 6) has black text on dark grey background. This unfortunately makes using the app very difficult, especially in daylight situations.

Screenshot:

[image]

bennesp, 2018-09-25:
Maybe a OnePlus related thing so. Just to be clear, the above bug seems to happen even changing the Oneplus6 android theme (Dark/Light/Colors).  If there is something I can do to help in the debugging, please let me know.

ikarus23, 2018-09-25:
Intersting. How does this testing version of MCT (with Material Theme) work out for you? http://tests.icaria.de/MifareClassicTool-2.2.3-testing4.apk

bennesp, 2018-09-25:
This version is working perfectly 🤩

ikarus23, 2018-09-25:
Great! I pushed to the master branch so it will be part of the next release.

ikarus23, 2018-09-27:
I had to change some small things. Could you please try [this testing version](http://tests.icaria.de/MifareClassicTool-2.2.3-testing5.apk) and see if it is still working properly? Thanks!

bennesp, 2018-09-27:
Just to let you know, I have just discovered that the problem was appearing
also in Android 6 in a P8 Lite.
But the last apk is working perfectly 👍
Thank you for your time

ikarus23, 2018-09-27:
Thank you too. I noticed this bug too on a Samsung S9. Really strange. Well I guess it's working now. ;)

closed 2018-09-27 by ikarus23