ikarus23/MifareClassicTool #203: Xiaomi MI 8 compatibility on MUIU 10.0.2.0

agsola, 2018-10-04:
I've just installed Mifare Classic Tool in this mobile.
I can read the tags with no problem, but when trying to clone from a dump, I get this error:
"Error. Some error ocurred while writing. This could be really bad"
I've tried cloning two different source tags to several different target tags (clonable tags).

If I disable "Enable writing to manufacturer block" it works.
I'm also able to properly do a "Factory Format", which finishes just fine.

These (https://aliexpress.com/item/5pcs-UID-IC-card-Changeable-Smart-Keyfobs-Key-Tags-Card-for-1K-S50-MF1-libnfc-RFID/32788592746.html) are the tags I bought, which state that are compatible with MCT.
Any idea or help?

ikarus23, 2018-10-04:
Although the tags are marketed as "MCT compatible" I don't think they are. It happend to me too. For a lot of tags on Aliexpress the description is just wrong (copied from another shop). You can see in the comment section that someone has the exact same issue as you do. He is says it does not work with MCT.

Do you have a Proxmark3 or a libnfc compatible RFID reader like the ACR122u?

ikarus23, 2018-10-04:
[These tags should work](https://de.aliexpress.com/item/5pcs-lot-Android-App-MCT-Modify-UID-Changeable-NFC-1k-s50-13-56MHz-keyfob-Block-0/32750649140.html?spm=a2g0x.search0104.3.51.493b5105OGW9j6&ws_ab_test=searchweb0_0,searchweb201602_2_10320_10065_10068_10547_318_10548_10696_10084_10083_10618_452_10304_10307_10820_532_10821_10302_204_10843_10059_10884_10887_100031_10319_10103,searchweb201603_60,ppcSwitch_0&algo_expid=eaf7b67c-cf64-4321-9f1f-028c72d6a561-6&algo_pvid=eaf7b67c-cf64-4321-9f1f-028c72d6a561&priceBeautifyAB=0). At some users claim they used them successfully with MCT.

agsola, 2018-10-05:
Thank you!

Unfortunately I don't have any of those.
I'll try to buy new ones and try.

closed 2018-10-26 by ikarus23

agsola, 2018-10-26:
Just in case you wanna know, I received yesterday my new tags. And they work perfectly in the Mi 8 (if you want to add to the compatibility list).

ikarus23, 2018-10-26:
That is great news. Thanks for the feedback!