ikarus23/MifareClassicTool #204: Application is stopping on Amazfit SmartWatch

TingyiChen, 2018-10-06:
Log:`D/AndroidRuntime( 1675): Shutting down VM
--------- beginning of crash
E/AndroidRuntime( 1675): FATAL EXCEPTION: main
E/AndroidRuntime( 1675): Process: de.syss.MifareClassicTool, PID: 1675
E/AndroidRuntime( 1675): java.lang.RuntimeException: Unable to start activity ComponentInfo{de.syss.MifareClassicTool/de.syss.MifareClassicTool.Activities.MainMenu}: android.util.AndroidRuntimeException: You cannot combine swipe dismissal and the action bar.
E/AndroidRuntime( 1675):        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2325)
E/AndroidRuntime( 1675):        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2387)
E/AndroidRuntime( 1675):        at android.app.ActivityThread.access$800(ActivityThread.java:151)
E/AndroidRuntime( 1675):        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1303)
E/AndroidRuntime( 1675):        at android.os.Handler.dispatchMessage(Handler.java:102)
E/AndroidRuntime( 1675):        at android.os.Looper.loop(Looper.java:135)
E/AndroidRuntime( 1675):        at android.app.ActivityThread.main(ActivityThread.java:5254)
E/AndroidRuntime( 1675):        at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime( 1675):        at java.lang.reflect.Method.invoke(Method.java:372)
E/AndroidRuntime( 1675):        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
E/AndroidRuntime( 1675):        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)
E/AndroidRuntime( 1675): Caused by: android.util.AndroidRuntimeException: You cannot combine swipe dismissal and the action bar.
E/AndroidRuntime( 1675):        at com.android.internal.policy.impl.PhoneWindow.requestFeature(PhoneWindow.java:333)
E/AndroidRuntime( 1675):        at com.android.internal.policy.impl.PhoneWindow.generateLayout(PhoneWindow.java:3410)
E/AndroidRuntime( 1675):        at com.android.internal.policy.impl.PhoneWindow.installDecor(PhoneWindow.java:3725)
E/AndroidRuntime( 1675):        at com.android.internal.policy.impl.PhoneWindow.setContentView(PhoneWindow.java:401)
E/AndroidRuntime( 1675):        at android.app.Activity.setContentView(Activity.java:2145)
E/AndroidRuntime( 1675):        at de.syss.MifareClassicTool.Activities.MainMenu.onCreate(MainMenu.java:109)
E/AndroidRuntime( 1675):        at android.app.Activity.performCreate(Activity.java:5990)
E/AndroidRuntime( 1675):        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1106)
E/AndroidRuntime( 1675):        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2278)
E/AndroidRuntime( 1675):        ... 10 more
W/ActivityManager(  384):   Force finishing activity 1 de.syss.MifareClassicTool/.Activities.MainMenu
W/InputMethodManagerService(  384): Window already focused, ignoring focus gain of: com.android.internal.view.IInputMethodClient$Stub$Proxy@7cb78df attribute=null, token = android.os.BinderProxy@10aaf618
D/PowerManager(  384): HM_acquireLocked pkg: android tag: WakeLock (ActivityManager-Launch)
W/ActivityManager(  384): Activity pause timeout for ActivityRecord{16bbe783 u0 de.syss.MifareClassicTool/.Activities.MainMenu t181 f}`

After I reverted this https://github.com/ikarus23/MifareClassicTool/commit/02d85daa8c93140802d545b97567f7c47f91e80c
it works fine...

ikarus23, 2018-10-07:
An with the https://github.com/ikarus23/MifareClassicTool/commit/5f17a143cf78b629e20ede8ef6c92d0381b1e514 it did not work?

TingyiChen, 2018-10-07:
Not working..app stopped and gave error log above

ikarus23, 2018-10-08:
Ok, I will look into it. Side question: can you even operate the app properly from a smartwatch? I never tested it on Android Wear...

ikarus23, 2018-10-08:
After some research I'm very unsure about this one. Normally an app is developed for smart watches or for mobile devices (phones and tables). But both at the same time? Is that watch running a "normal" Android? As far as I know I don't target Android Wear OS devices. How did you install it? And were you able to use it properly (with the version that did not crash)?

TingyiChen, 2018-10-09:
The app menu is showing fine on my watch,and the mifare funcs works as my another phone with NFC..Only issue is that line causing crash...

TingyiChen, 2018-10-09:
My watch enabled adb by default when connect to computer so I can install apps on jt

ikarus23, 2018-10-09:
Interesting. Never thought MCT would work on something like a smart watch. However, I don't want to change the line back that is causing the issue. It would be best if there is a way for Android to detect if it is running on a watch and then use a different styles.xml. I will look into it if I find some time...

TingyiChen, 2018-10-10:
Same..Hope can find a solution for this...I don't know if same issue happening on other smart watch...

closed 2018-10-10 by TingyiChen