ikarus23/MifareClassicTool #206: Moto G5s Plus

Yorixiriamori, 2018-10-16:
Hi, I'm a big fan of your work.

Thank you for your commitment to improving the application.

I have a question about how to use the application on Motorola Moto G5s Plus.

Apparently the Moto G5s Plus uses the same NFC chip as the Moto G5 plus (I read that it works), but when trying to read a tag, even if all the passwords are the same, some passwords (when reading the tag) appear blank as if had not been found.
Sometimes it returns as if it did not find industry passwords and was not displayed.

I think it's a mistake, because if I read it again, it reads the password it did not read and stops reading another password.

I apologize for English if there is an error and, from now on, thank you very much.

ikarus23, 2018-10-18:
Hi! Thank you for your kind words.

It looks like the Moto G5s Plus has the same issue as my Nexus 5X (https://github.com/ikarus23/MifareClassicTool/issues/106). It's most likely an error in Android. However, you can go to the settings of MCT and enable "retry authentication" with something like 5 to 10. I hope this helps.

Yorixiriamori, 2018-10-18:
Thank you very much.

I made that change and it started working, but something funny happened, I have a blank tag that the passwords are possibly fff ... or 000 .... When reading now, I can read with both passwords and it appears that the passwords are 000 or fff, depending on the password I use to read.

I'm sending two prints.

[image]
[image]

ikarus23, 2018-10-22:
Looks like you have another well known bug. However, until now I was not able to reproduce it with one of my devices. Have a look at https://github.com/ikarus23/MifareClassicTool/issues/66.

ikarus23, 2018-12-28:
Closed as duplicate of https://github.com/ikarus23/MifareClassicTool/issues/66. Let's continue the discussion over there, if there are any news.

closed 2018-12-28 by ikarus23