ikarus23/MifareClassicTool #207: UI freeze while mapping/reading [Google Pixel, Android 9]

ikarus23, 2018-10-18:
A lot of users complaining that MCT freezes on Googles Pixel with Android 9. This is the stacktrace:
```

"main" tid=1 Native
"main" prio=5 tid=1 Native
  | group="main" sCount=1 dsCount=0 flags=1 obj=0x74eeca78 self=0x720c614c00
  | sysTid=26192 nice=-10 cgrp=default sched=0/0 handle=0x729202f548
  | state=S schedstat=( 9638645976 816419133 23733 ) utm=749 stm=214 core=3 HZ=100
  | stack=0x7ff9657000-0x7ff9659000 stackSize=8MB
  | held mutexes=
  #00  pc 0000000000070d7c  /system/lib64/libc.so (__ioctl+4)
  #01  pc 00000000000292dc  /system/lib64/libc.so (ioctl+144)
  #02  pc 000000000005a740  /system/lib64/libbinder.so (android::IPCThreadState::talkWithDriver(bool)+260)
  #03  pc 000000000005b4e0  /system/lib64/libbinder.so (android::IPCThreadState::waitForResponse(android::Parcel*, int*)+60)
  #04  pc 000000000005b32c  /system/lib64/libbinder.so (android::IPCThreadState::transact(int, unsigned int, android::Parcel const&, android::Parcel*, unsigned int)+176)
  #05  pc 000000000005130c  /system/lib64/libbinder.so (android::BpBinder::transact(unsigned int, android::Parcel const&, android::Parcel*, unsigned int)+72) 
  #06  pc 000000000012fe70  /system/lib64/libandroid_runtime.so (android_os_BinderProxy_transact(_JNIEnv*, _jobject*, int, _jobject*, _jobject*, int)+152)
  at android.os.BinderProxy.transactNative (Native method)
  at android.os.BinderProxy.transact (Binder.java:1127)
  at android.nfc.INfcTag$Stub$Proxy.reconnect (INfcTag.java:307)
  at android.nfc.tech.BasicTagTechnology.close (BasicTagTechnology.java:123)
  at android.nfc.tech.MifareClassic.close (MifareClassic.java:73)
  at de.syss.MifareClassicTool.MCReader.close (MCReader.java:1068)
  at de.syss.MifareClassicTool.Common.checkForTagAndCreateReader (Common.java:844)
  at de.syss.MifareClassicTool.Activities.KeyMapCreator.onCreateKeyMap (KeyMapCreator.java:388)
  at java.lang.reflect.Method.invoke (Native method)
  at android.view.View$DeclaredOnClickListener.onClick (View.java:5629)
  at android.view.View.performClick (View.java:6597)
  at android.view.View.performClickInternal (View.java:6574)
  at android.view.View.access$3100 (View.java:778)
  at android.view.View$PerformClick.run (View.java:25885)
  at android.os.Handler.handleCallback (Handler.java:873)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at android.os.Looper.loop (Looper.java:193)
  at android.app.ActivityThread.main (ActivityThread.java:6669)
  at java.lang.reflect.Method.invoke (Native method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:493)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:858)
```
At the moment, I'm not sure what exactly is causing this issue.

josecurioso, 2018-10-26:
I have this exact issue on my Pixel XL running Android 9. I have been trying to figure it out using the Android Studio Debugger but I'm pretty new to android development and am completely lost. Sorry I can't be of any more help but if you need to test anything and don't have the Pixel at hand I will gladly help out.

ikarus23, 2018-10-26:
Hi! Thanks for your feedback. So MCT works on Pixel 2 devices but not on the original Pixel (regardless of XL or not).

Can you try this [testing version of MCT](http://tests.icaria.de/MifareClassicTool-2.2.4-testing2.apk)?

josecurioso, 2018-10-26:
So after uninstalling the other version, rebooting and installing this one it still does not work but the app doesn't freeze anymore.
Steps:
-When I approach the tag the "New tag found" Toast pops up.
-Then when I click "Start mapping and read tag" it waits a second and then another message informs me that there is no Mifare Classic card (but there is)
-After that if I separate the card and try again the "New tag found" message doesn't appear again and I have to deactivate/activate NFC for it to work again.

Here is the logcat extract from the moment I approach the tag: https://pastebin.com/WmC5UsX7

ikarus23, 2018-11-08:
Thanks for the feedback.
Can you try [this testing version](http://tests.icaria.de/MifareClassicTool-2.2.4-testing3.apk)?

josecurioso, 2018-11-08:
With this version, when I approach the tag the app hangs with the "start mapping and read tag" button pressed and after a while android prompts me to either keep waiting or close the app.
Here are the messages logged when I approached the tag: https://pastebin.com/s7kMFN34

josecurioso, 2019-04-26:
Any news regarding this issue? After the last security updates it still happens on Pie and when Q is released I can try again.

josecurioso, 2019-05-02:
Just installed the Android Q Beta 2 on my Pixel XL and I'm happy to report that for some reason everything works now. It doesn't hang and works just like back in Oreo. Keep in mind that Q is still in beta and is a bit rough around the edges.

ikarus23, 2019-05-03:
Thanks for testing and the great news! I was hoping for something like this, because until now I did not find any workaround for this issue. Google fixing its NFC stack/driver sounds like the best solution for me (no work for me :) )

HarvexARC, 2019-07-26:
Testing 20190726 :
Pixel XL Gen1 running android 9 with ver 2.3.1 GooglePlay version.
Still got freeze when "Start mapping and read tag"

ikarus23, 2020-08-16:
Since @josecurioso reports the issue was fixed with Android 10, I will close this issue. It looks like this was a Google ROM issue, and not an issue related to MCT.

closed 2020-08-16 by ikarus23