ikarus23/MifareClassicTool #209: MIUI could read a card but LineageOS didn't

wbsdty331, 2018-10-25:
Hi ikarus,
I am using Xiaomi 5(gemini),When I use MIUI 10, use MCT to read or write a JCOP card have no problem, just sometimes MCT Notice it's not Mifare Classic Tags, But sometimes it works.

But I flashed LineageOS 15.1, no matter what i do, MCT said "Not Mifare Classic Tags",I don't know why.

Thanks.

ikarus23, 2018-10-26:
Hi!
Hmm... a JCOP card in Mifare Classic emulation mode can have different SAK values, for example. This might trick Android (or MCT?!) into thinking it is not a Mifare Classic tag.

However, I have no clue why it worked with MIUI sometimes. Maybe this operating system handles this tag different (e. g. because of its SAK value).

wbsdty331, 2018-10-27:
I don't know why :(
Should I provide some log?

wbsdty331, 2018-12-06:
Or How to compare these ROM SAK Value?

ikarus23, 2018-12-06:
Sorry for not answering earlier. I somehow overlooked your message.
Unfortunately, I don't know what to do. Have you checked other apps? Are they able of reading/writing Mifare Classic?

wbsdty331, 2018-12-07:
I tried use TWRP to switch system between MIUI and Lineage OS, use a same card.
When I done, I will upload some information.
 
BTW, Where is log?

wbsdty331, 2018-12-07:
same as NFC Tools
[image]
[image]

Sometimes 0x20 but sometimes 0x28.

ikarus23, 2018-12-09:
Well, this is interesting. This proves something is wrong with the OS (drivers) or the tag itself (and not the MCT app). Do you have access to another Android phone? You could try another device to find out if the tag is faulty or the OS.

wbsdty331, 2018-12-10:
Tested my roommate's Huawei Honor 10
Like MIUI but not test on Lineage

ikarus23, 2018-12-10:
It has the same strange behavior with a different OS? I'm starting to believe there is something wrong with the tag.

wbsdty331, 2018-12-11:
yes.

ikarus23, 2018-12-28:
I will close this issue since it seems it is not related to MCT but the tag or the OS. Feel free to leave any feedback if you find out what exactly is causing this issue.

closed 2018-12-28 by ikarus23