ikarus23/MifareClassicTool #211: 4k -> 1K - Trimmed clone question

sibero80, 2018-10-26:
Hello there! greetings from Colombia.

**Here's my situation** (I'm by no means a programmer, more like a scrip kiddie)

1. I'm able to perform a full dump (known keys) of a 4k mifare card but using my phone (with this app) and my linux machine (libnfc)
2. Using the app, I'm able to write the dump into a 1k card (trimming sectors 16-39).
3. The 1k cloned card works.
4. When I compare the dumps (original 4k vs trimmed 1K) sector 0 and sector 15 differ slightly.
5. When I try to clone the cloned 1k card with the Linux machine (miLazyCracker) if dumps successfully, but when trying to write into another 1k tag, I get an error regarding SAK code.

**So here are my questions:**
1. Is there a better way to trim clone a 4k tag dump into a 1k tag? (when the remaining sectors are not used).
2. How could the APP dump txt files be converted into a LIBNFC compatible MFD dump file?.
3. How could the APP ingest/import LIBNFC mfd dump files?.
4. Wishful thining: Could miLazyCracker libraries be included on this app? it's performance is greater than MFOC and MCFUK and could actually be practical.

Thank you!

ikarus23, 2018-10-26:
Hi!

Your situation:

4. Just out of curiosity, can you show me what differs in the dump? I guess sector 0 block 0 is different. But what is different in sector 15?
5. Hmmm maybe it is because the tag is different. Different tags (e. g. in size or type) have different SAK values.

Your questions:

1. I don't know if it is better, but you can just trim an `.mdf` file from 4k to 1k and write it.
2. You can convert the files with the scripts in the [dump-file-converters](https://github.com/ikarus23/MifareClassicTool/tree/master/tools/dump-file-converter) folder. Use the `mct2eml.py` first and then the `eml2mfd.py`.
3. Same way as 2. Use `mfd2eml.py` and `eml2mct.py`. Then place the file in `sdcard/MifareClassicTool/dump-files/`
4. No, cracking Mifare Classic using Androids NFC API is not possible. Even on a rooted phone you will need customized drivers for your NFC chip.

ikarus23, 2018-12-28:
Since there are no more questions I'm closing this issue. I hope you got everything to work the way you want.

closed 2018-12-28 by ikarus23