ikarus23/MifareClassicTool #220: Weird key file behaviour

jaymon1998, 2018-12-19:
I'm having some issues reading the mifare classic 1k card with the key files. 

1. The keys (A & B) of all the sectors are FFFFFFFFFFFF. Initially I used the std.keys file containing the key to read the card. I got the error saying "no valid key found in chosen key file" but if i try 2 or more times, it is able to read the card and shows all the sectors. I tried using the MCT to create a key file with only FFFFFFFFFFFF as the key. It works normally, able to read the card. Then, I add another key to the file say A0A1A2A3A4A5, sometimes it reads on first mapping or I need to try 2-3 times.

2. I changed the key A (all access conditions remain as transport configuration) of sector 14 to A0A1A2A3A4A5. When I read the card with Key FFFFFFFFFFF **only**, all sectors are shown except 14 and 15. Turns out that if i change the key A of one sector, the subsequent sectors below it shows "No keys found (or dead sector). If I read only sector 14 with key A0A1A2A3A4A5, it is able to show. I've tried putting the two keys in one file as well as in two files individually, I need to press the start mapping button multiple times and still, only sectors with key FFFFFFFFFFFF are shown.

3. I also changed the key A of sector 15 to A0A1A2A3A4A5 and read only sector 14-15 with that key. Only sector 14 is shown. If i read sector 15 alone, the sector is shown as well.

I'm new to android development and Mifare Classic and I'm confused about these. I've read similar issues here about the "no valid key found" and the problem seems to come from the device. I'm using Centerm K9 device (It's a portable payment terminal) and it runs on Android 5.1

Piket564, 2018-12-19:
I think also that.

> I've read similar issues here about the "no valid key found" and the problem seems to come from the device.

Maybe is the device.
Whit MCT i've never had any problem

ikarus23, 2018-12-20:
Hi, thank you for your very detailed tests. However, I think @Piket564 might be right. It really sounds like an issue of the device or the tag. Have you tried to read the tag with another device? Have you tried to read another Mifare Classic tag with your Centerm K9?

Also, you can try the "retry authentication" option? Maybe it helps. At least for reading tags.

jaymon1998, 2018-12-21:
I couldn't get my hand on another device (will keep trying) and the "retry authentication" options doesn't help either. Here's what I did.

All sectors are authenticated with Key A(1) of FFFFFFFFFFFF. Only one sector, say Sector 14, uses a different Key A(2). So, when Sector 14 fails to authenticate with Key A(1), the next Sector 15 gets authenticated with Key A(1), the same key A, but it failed. Therefore, I close() the mfc connection and connect() it again before authenticating sector 15, it works....

for (int j = 0; j < secCount; j++) {
                        for(int i = 0; i< KEYS.length; i++){
                            if(mfc.authenticateSectorWithKeyA(j, KEYS[i])){ 
                                isAuthenticated = true;
                                break;
                            } else { 
                                mfc.close();
                                mfc.connect();
                            }
                        }
                        if (isAuthenticated) {
                           //the readblock() code

ikarus23, 2018-12-21:
Interesting workaround you fond there. I don't think it is very advisable because the mapping process should take longer this way. However, maybe we should make this an option. If I recall this right, there are more users out there with the same or a similar issue.

ikarus23, 2018-12-21:
Can you try [this testing version](http://tests.icaria.de/MifareClassicTool-2.2.4-testing4.apk)? I did implement your workaround at different section in the code. Does it still work for you? What happens if you remove the tag during mapping/reading?

jaymon1998, 2018-12-23:
The testing version still produces the same issue. If I remove the tag during mapping, it still completes the mapping process (visually based on the progress bar) and proceed to the next activity but just showing the "Reading tag...(Don't remove tag)" screen. It doesn't shows the sectors or error message, although pressing back do return to the main menu. 

> I don't think it is very advisable because the mapping process should take longer this > 

I do know that, but I think that won't be a major issue (as of now) in the application I'm implementing this as I won't be reading/writing all the sectors and I'm only told to work with MC 1k only.

jaymon1998, 2018-12-28:
Another issue regarding the sector trailer:
Key A: `A0A1A2A3A4A5`
AC: `A1EF05(69)`
Key B: `FFFFFFFFFFFF` (shown in dashes '-' due to access condition)

Based on the AC for Block 2:-
>**Location:_____Read:________Write:______Incr:________Decr./Trans/Rest:**
Block 2:______Key A|B______Key B______Key B________Key A|B

1. If i write with Key B `(FFFFFFFFFFFF),` error shows "No valid key found in chosen key file".
1. If i write with Key A `(A0A1A2A3A4A5),` error shows "Block is read only or not writable with keys from key file".

Same issue for increment and decrement:

3. If i write with Key B `(FFFFFFFFFFFF),` error shows "No valid key found in chosen key file".
4. If i write with Key A `(A0A1A2A3A4A5),` error shows "Not a value block, insufficient rights, or incorrect value" (block 2 **is** a Value Block).

But when I implement my own code in my application, it worked as intended, writing with Key B. This is confusing. I should really test it on another device.

ikarus23, 2018-12-28:
Well, this is indeed strange. At least parts of it.
Regarding 1: This fails the same way as it fails for reading. MCT tries to authenticate with KeyB before reading/writing. Since this failed for reading (this is why it it shown as "---" and not because of the AC) it will most likely fail for writing too.
Regarding 2: Write *must* fails because KeyA is not allowed to write.
Regarding 3: Again, like 1, this should work. But I think MCT isn't trying to inc/dec because the authentication in the key mapping process fails.
Regarding 4: Increment *must* fail because of the ACs, but again, decrement should work.

It seams like everything is connected to the fact, that the authentication fails sometimes (when the reader was not reconnected before the next try).

ikarus23, 2018-12-28:
Does [this testing version](http://tests.icaria.de/MifareClassicTool-2.2.4-testing5.apk) help?

jaymon1998, 2019-01-01:
it doesn't seems to work

ikarus23, 2020-08-16:
Sorry, I still don't have any clue why your device does not work as intended and why my workarounds did not fix anything.

The fix to `close()` and `connect()` directly after a failed authentication is even stranger. As the [official documentation states](https://developer.android.com/reference/android/nfc/tech/MifareClassic#authenticateSectorWithKeyA(int,%20byte[])):
> A failed authentication attempt causes an implicit reconnection to the tag, so authentication to other sectors will be lost.

So there should be no need to do this. Maybe it is somehow related to strange timing issues. But I'm running out of ideas. And without debug access to a Centerm K9 I don't think there is much more I can do. I'm sorry.

I will close this issue for now. Feel free to report back and to reopen it, if you have any idea on how to go on from here.

closed 2020-08-16 by ikarus23