ikarus23/MifareClassicTool #221: cant write UID to Chinese magic card.

junjun4747, 2018-12-19:
im a newbie, in order to clone my card, i have bought many useless things.
first was NTAG cards.  the seller marked the product with universal logo, so i thought it would be perfect. then i learned about different cards ( classic , ultralight ect.) second purchase was classic ( tags stickers and cards). third purchase was chinese magic cards. now i know there are 2 kinds of magic cards. direct writable and ones with special commands.
if i cant write uid with my phone, does that mean i have the one that requires special commands? is there a way to check which card i have without a usb reader? my reader is still on the way to me.

Piket564, 2018-12-19:
I don't think that is an issue of MCT.
Try that to have NFC card info.

[https://play.google.com/store/apps/details?id=com.wakdev.wdnfc&hl=en](NFC INFO)

kgamecarter, 2018-12-20:
There is no app can check chinese magic card back door command.
That is the restrictions of the Android system.
You can only wait for the card reader to check.

ikarus23, 2018-12-20:
@kgamecarter  is right. It is most likely you have the the UID changeable tags, that require special commands. But you can only check with an USB reader or a Proxmark3. Android (and therefore MCT) can only change the UID of tags that need no special command. Unfortunately, the shop descriptions on Aliexpress or ebay are often wrong. Before buying, check the feedback section and find out if someone successfully used them on Android. [These](https://de.aliexpress.com/item/5pcs-lot-Android-App-MCT-Modify-UID-Changeable-NFC-1k-s50-13-56MHz-keyfob-Block-0/32750649140.html?spm=a2g0x.search0104.3.51.493b5105OGW9j6&ws_ab_test=searchweb0_0,searchweb201602_2_10320_10065_10068_10547_318_10548_10696_10084_10083_10618_452_10304_10307_10820_532_10821_10302_204_10843_10059_10884_10887_100031_10319_10103,searchweb201603_60,ppcSwitch_0&algo_expid=eaf7b67c-cf64-4321-9f1f-028c72d6a561-6&algo_pvid=eaf7b67c-cf64-4321-9f1f-028c72d6a561&priceBeautifyAB=0) or [these](https://de.aliexpress.com/item/100pcs-lot-Android-App-MCT-Modify-UID-Changeable-NFC-1k-s50-13-56MHz-card-Block-0/32757062972.html?spm=a2g0x.search0104.3.2.150e10a6noihYK&ws_ab_test=searchweb0_0,searchweb201602_2_10320_10065_10068_10547_319_317_10548_10696_453_10084_454_10083_10618_10304_10307_10820_10821_538_537_10302_536_10843_10059_10884_10887_100031_10319_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=024e85b2-6cc4-4439-915e-130dc36bacc3-0&algo_pvid=024e85b2-6cc4-4439-915e-130dc36bacc3) tags should work.