ikarus23/MifareClassicTool #224: No key found or bad sector

dimitrire, 2019-01-08:
HI, I have a mifare card and the problem is that I can not read sector 1 with MCT on Android, how can I find the key? I also have the ACR122u reader, thank you for your help, Regards Dimitri

dimitrire, 2019-01-08:
[image]

dimitrire, 2019-01-08:
The détails of the card

ikarus23, 2019-01-15:
Hi, have a look at the [getting started](https://github.com/ikarus23/MifareClassicTool#getting-started) section of the readme. there are the names of 2 tools that can help you. Also check out [miLazyCracker](https://github.com/ilumitr/miLazyCracker). This tool should be able to do everything for you.

dimitrire, 2019-01-19:
Thank you for your answer, I'm under Kali, is this a good version? I will try to install miLazyCracker is you make a return if I get there, thanks again.

ikarus23, 2019-01-20:
You are welcome. And Kali should be fine. mfcuk and mfoc are even preinstalled. Maybe mfoc is enough to recover the key. But you can try miLazyCracker as well.

dimitrire, 2019-01-20:
Mfoc is not enough, there are 5 sectors all the keys are ffffffffffff except for the sector 1 where it does not find there key A nor B, I tried to install miLazyCracker but after I do not know what write as command or if it installs without error

ikarus23, 2019-01-20:
mfoc does not only check if the keys are `ffffffffffff`, it also tries to recover unknown keys. However, if you have a tag with a hardened PRNG mfoc will not work.

I can't help you with myLazyCracker. I never used it. I've heard from people that it works fine. According to the readme, you just have to run it. Maybe watching the [presentation of the creator](https://www.youtube.com/watch?v=VcU3Yf5AqQI) will help you.

dimitrire, 2019-01-27:
root@kali:~/miLazyCracker# ./miLazyCrackerFreshInstall.sh
+ sudo apt-get install git libnfc-bin autoconf libnfc-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
autoconf is already the newest version (2.69-11).
git is already the newest version (1:2.20.1-1).
libnfc-bin is already the newest version (1.7.1-4+b1).
libnfc-dev is already the newest version (1.7.1-4+b1).
The following packages were automatically installed and are no longer required:
  libgtk2-perl libpango-perl libperl5.26
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 454 not upgraded.
+ '[' -d mfoc ']'
+ git clone https://github.com/nfc-tools/mfoc.git
Cloning into 'mfoc'...
remote: Enumerating objects: 526, done.
remote: Total 526 (delta 0), reused 0 (delta 0), pack-reused 526
Receiving objects: 100% (526/526), 230.11 KiB | 382.00 KiB/s, done.
Resolving deltas: 100% (330/330), done.
+ cd mfoc
+ git reset --hard
HEAD is now at ba072f1 update debian dir with up-to-date packaging
+ git clean -dfx
+ patch -p1
patching file src/mfoc.c
Hunk #1 succeeded at 72 with fuzz 2 (offset 14 lines).
Hunk #2 FAILED at 159.
Hunk #3 succeeded at 257 (offset 15 lines).
Hunk #4 FAILED at 471.
Hunk #5 FAILED at 554.
Hunk #6 FAILED at 770.
Hunk #7 FAILED at 1029.
5 out of 7 hunks FAILED -- saving rejects to file src/mfoc.c.rej
+ patch -p1
patching file src/mfoc.c
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file src/mfoc.c.rej
+ patch -p1
patching file src/mfoc.c
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
2 out of 2 hunks ignored -- saving rejects to file src/mfoc.c.rej
+ patch -p1
patching file src/mfoc.c
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
6 out of 6 hunks ignored -- saving rejects to file src/mfoc.c.rej
patching file src/mfoc.h
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] n
Skipping patch.
3 out of 3 hunks ignored -- saving rejects to file src/mfoc.h.rej
+ autoreconf -vfi
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force 
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
configure.ac:17: error: possibly undefined macro: AC_MSG_ERROR
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1
+ ./configure
configure: error: cannot find install-sh, install.sh, or shtool in "." "./.." "./../.."
+ make
make: *** No targets specified and no makefile found.  Stop.
+ sudo make install
make: *** No rule to make target 'install'.  Stop.
+ '[' -d crypto1_bs ']'
+ git clone https://github.com/aczid/crypto1_bs
Cloning into 'crypto1_bs'...
remote: Enumerating objects: 368, done.
remote: Total 368 (delta 0), reused 0 (delta 0), pack-reused 368
Receiving objects: 100% (368/368), 132.85 KiB | 425.00 KiB/s, done.
Resolving deltas: 100% (237/237), done.
+ cd crypto1_bs
+ git reset --hard
HEAD is now at 89de1ba Merge pull request #28 from dkgitdev/patch-1
+ git clean -dfx
+ patch -p1
patching file libnfc_crypto1_crack.c
Hunk #1 succeeded at 730 with fuzz 2 (offset 17 lines).
+ make get_craptev1
wget http://crapto1.netgarage.org/craptev1-v1.1.tar.xz
--2019-01-27 14:47:19--  http://crapto1.netgarage.org/craptev1-v1.1.tar.xz
Resolving crapto1.netgarage.org (crapto1.netgarage.org)... 176.9.4.150
Connecting to crapto1.netgarage.org (crapto1.netgarage.org)|176.9.4.150|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-01-27 14:47:19 ERROR 404: Not Found.

make: *** [Makefile:34: get_craptev1] Error 8
+ make get_crapto1
wget http://crapto1.netgarage.org/crapto1-v3.3.tar.xz
--2019-01-27 14:47:19--  http://crapto1.netgarage.org/crapto1-v3.3.tar.xz
Resolving crapto1.netgarage.org (crapto1.netgarage.org)... 176.9.4.150
Connecting to crapto1.netgarage.org (crapto1.netgarage.org)|176.9.4.150|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-01-27 14:47:19 ERROR 404: Not Found.

make: *** [Makefile:38: get_crapto1] Error 8
+ make
gcc -std=gnu99 -O3 -march=native solve_bs.c crypto1_bs.c crypto1_bs_crack.c  crapto1-v3.3/crapto1.c crapto1-v3.3/crypto1.c -I crapto1-v3.3/  craptev1-v1.1/craptev1.c -I craptev1-v1.1/ -o solve_bs -lpthread -lm
gcc: error: crapto1-v3.3/crapto1.c: No such file or directory
gcc: error: crapto1-v3.3/crypto1.c: No such file or directory
gcc: error: craptev1-v1.1/craptev1.c: No such file or directory
make: *** [Makefile:19: solve_bs] Error 1
+ sudo cp -a libnfc_crypto1_crack /usr/local/bin
cp: cannot stat 'libnfc_crypto1_crack': No such file or directory
+ sudo cp -a miLazyCracker.sh /usr/local/bin/miLazyCracker
+ echo Done.
Done.
root@kali:~/miLazyCracker# ./miLazyCracker.sh
./miLazyCracker.sh: line 52: mfoc: command not found

dimitrire, 2019-01-27:
Hello , here are the mistakes I have

ikarus23, 2019-01-27:
Hi. There are a lot of issues here.
* Applying the patch to mfoc.c failed
* autoreconf failed --> ./configure failed --> make failed
* The file `http://crapto1.netgarage.org/craptev1-v1.1.tar.xz` no longer exists
* The file `http://crapto1.netgarage.org/crapto1-v3.3.tar.xz` no longer exists

There is no general solution to that. You have to work it out your self. Maybe it is best you try mfoc first. It should be easier to install. Maybe it does get the job done.

dimitrire, 2019-01-28:
mfcuk - 0.3.8
Mifare Classic DarkSide Key Recovery Tool - 0.3
by Andrei Costin, zveriu@gmail.com, http://andreicostin.com

INFO: Connected to NFC reader: ACS / ACR122U PICC Interface

VERIFY: 
	Key A sectors: 0Segmentation fault

root@kali:~# mfoc -P 3 -O /dev/null
Found Mifare Classic Mini tag
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
* UID size: single
* bit frame anticollision supported
       UID (NFCID1): 3b  c9  f5  a9  
      SAK (SEL_RES): 09  
* Not compliant with ISO/IEC 14443-4
* Not compliant with ISO/IEC 18092

Fingerprinting based on MIFARE type Identification Procedure:
* MIFARE Mini 0.3K
* SmartMX with MIFARE 1K emulation
Other possible matches based on ATQA & SAK values:

Try to authenticate to all sectors with default keys...
Symbols: '.' no key found, '/' A key found, '\' B key found, 'x' both keys found
[Key: ffffffffffff] -> [x.xxx]
[Key: a0a1a2a3a4a5] -> [x.xxx]
[Key: d3f7d3f7d3f7] -> [x.xxx]
[Key: 000000000000] -> [x.xxx]
[Key: b0b1b2b3b4b5] -> [x.xxx]
[Key: 4d3a99c351dd] -> [x.xxx]
[Key: 1a982c7e459a] -> [x.xxx]
[Key: aabbccddeeff] -> [x.xxx]
[Key: 714c5c886e97] -> [x.xxx]
[Key: 587ee5f9350f] -> [x.xxx]
[Key: a0478cc39091] -> [x.xxx]
[Key: 533cb6c723f6] -> [x.xxx]
[Key: 8fd0a4f256e9] -> [x.xxx]

Sector 00 - Found   Key A: ffffffffffff Found   Key B: ffffffffffff
Sector 01 - Unknown Key A               Unknown Key B
Sector 02 - Found   Key A: ffffffffffff Found   Key B: ffffffffffff
Sector 03 - Found   Key A: ffffffffffff Found   Key B: ffffffffffff
Sector 04 - Found   Key A: ffffffffffff Found   Key B: ffffffffffff

Using sector 00 as an exploit sector
Card is not vulnerable to nested attack

dimitrire, 2019-01-28:
root@kali:~# cd miLazyCracker
root@kali:~/miLazyCracker# ./miLazyCracker.sh
Found Mifare Classic Mini tag
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04  
* UID size: single
* bit frame anticollision supported
       UID (NFCID1): 3b  c9  f5  a9  
      SAK (SEL_RES): 09  
* Not compliant with ISO/IEC 14443-4
* Not compliant with ISO/IEC 18092

Fingerprinting based on MIFARE type Identification Procedure:
* MIFARE Mini 0.3K
* SmartMX with MIFARE 1K emulation
Other possible matches based on ATQA & SAK values:

Try to authenticate to all sectors with default keys...
Symbols: '.' no key found, '/' A key found, '\' B key found, 'x' both keys found
[Key: ffffffffffff] -> [x.xxx]
[Key: a0a1a2a3a4a5] -> [x.xxx]
[Key: d3f7d3f7d3f7] -> [x.xxx]
[Key: 000000000000] -> [x.xxx]
[Key: b0b1b2b3b4b5] -> [x.xxx]
[Key: 4d3a99c351dd] -> [x.xxx]
[Key: 1a982c7e459a] -> [x.xxx]
[Key: aabbccddeeff] -> [x.xxx]
[Key: 714c5c886e97] -> [x.xxx]
[Key: 587ee5f9350f] -> [x.xxx]
[Key: a0478cc39091] -> [x.xxx]
[Key: 533cb6c723f6] -> [x.xxx]
[Key: 8fd0a4f256e9] -> [x.xxx]

Sector 00 - Found   Key A: ffffffffffff Found   Key B: ffffffffffff
Sector 01 - Unknown Key A               Unknown Key B
Sector 02 - Found   Key A: ffffffffffff Found   Key B: ffffffffffff
Sector 03 - Found   Key A: ffffffffffff Found   Key B: ffffffffffff
Sector 04 - Found   Key A: ffffffffffff Found   Key B: ffffffffffff

Using sector 00 as an exploit sector
Card is not vulnerable to nested attack
MFOC not possible, detected hardened Mifare Classic
Trying HardNested Attack...
libnfc_crypto1_crack ffffffffffff 16 B 4 B mfc_3bc9f5a9_foundKeys.txt
./miLazyCracker.sh: line 87: libnfc_crypto1_crack: command not found
MFOC not possible, detected hardened Mifare Classic
Trying HardNested Attack...
libnfc_crypto1_crack ffffffffffff 16 B 4 B mfc_3bc9f5a9_foundKeys.txt
./miLazyCracker.sh: line 87: libnfc_crypto1_crack: command not found
MFOC not possible, detected hardened Mifare Classic
Trying HardNested Attack...
libnfc_crypto1_crack ffffffffffff 16 B 4 B mfc_3bc9f5a9_foundKeys.txt
./miLazyCracker.sh: line 87: libnfc_crypto1_crack: command not found
MFOC not possible, detected hardened Mifare Classic
Trying HardNested Attack...
libnfc_crypto1_crack ffffffffffff 16 B 4 B mfc_3bc9f5a9_foundKeys.txt
./miLazyCracker.sh: line 87: libnfc_crypto1_crack: command not found
MFOC not possible, detected hardened Mifare Classic
Trying HardNested Attack...
libnfc_crypto1_crack ffffffffffff 16 B 4 B mfc_3bc9f5a9_foundKeys.txt
./miLazyCracker.sh: line 87: libnfc_crypto1_crack: command not found
MFOC not possible, detected hardened Mifare Classic

ikarus23, 2019-01-28:
Ok, so mfoc did not work because it is a hardened tag. And cracking the hardened tag did not work because `libnfc_crypto1_crack` was not installed correctly.

dimitrire, 2019-01-31:
how install  libnfc_crypto1_crack  ?

ikarus23, 2019-01-31:
I don't know. It looks like you have to install [crypto1_bs](https://github.com/aczid/crypto1_bs). But there issues because some code is missing. Have a look at https://github.com/aczid/crypto1_bs/issues/29. Maybe you get is to work (with some googleing).

Of course you can always by a Proxmark3. This device can crack the keys "out of the box".

closed 2019-03-31 by ikarus23