ikarus23/MifareClassicTool #225: I can not read card by acr122u without NFC.

kgamecarter, 2019-01-08:
I can use External NFC detect card. And my phone without NFC. 
But read and write button still disabled. 
[image]

ikarus23, 2019-01-15:
This really looks like a bug in MCT. But be warned: People report that External NFC is not working correctly (https://github.com/ikarus23/MifareClassicTool/issues/198 or https://github.com/ikarus23/MifareClassicTool/issues/123#issuecomment-388307001). I will look into it, but it might take some time.

ikarus23, 2019-01-27:
Is the External NFC service running correctly? I can't see the symbol at the top of your screenshot. And was there any error message? MCT tries to find out if External NFC is installed and running. If so, the buttons should be operable. I have no device without NFC right here, but I will borrow me one to check if I get the same behavior.

ikarus23, 2019-01-29:
Ok, I found the bug. MCT uses [getRunningServices(int)](https://developer.android.com/reference/android/app/ActivityManager.html#getRunningServices(int)) to find out if External NFC is running. `getRunningServices(int)` however is deprecated and no longer works. I will try to find a fix.

ikarus23, 2019-01-31:
I tested it with the device of a friend and it seem to work. Have you been able to test it?

kgamecarter, 2019-01-31:
Ok, Is's work now.
Thanks update.
[image]

ikarus23, 2019-02-01:
Great! The code is in the master branch and will be part of the next release.

closed 2019-02-01 by ikarus23