ikarus23/MifareClassicTool #226: Chinese Magic Card "Lost connection to the tag"

svedizz, 2019-01-16:
So my problem is that I can't read or write to my card anymore. It only worked once and ever since my first writing to the card I've been getting the "Lost connection to the tag" error message. Did my card break somehow, is it my phone, or is this an issue with the app? I'm using a Samsung Galaxy S7 Edge.

ikarus23, 2019-01-19:
Strange, but yes, it sounds like you card is broken. I can't tell what exactly caused the issue. However, since other users use MCT to write to Chinese Magic cards without any trouble, I think it might be a "bad" tag or the Galaxy S7 Edge. The S7 (Edge) just started to work with MIFARE Classic from Android 8.0 onwards. Maybe there are still issues.

svedizz, 2019-01-19:
I think I broke it somehow, because other non-CMC cards still works.

ikarus23, 2019-01-19:
Did writing to non-CMC cards work?

svedizz, 2019-01-19:
Yes, but obviously not to block 0 in sector 0.

ikarus23, 2019-01-19:
Indeed, very strange. Especially the fact that the card can be detected but reading/writing is not possible. Maybe it's just the proprietary MIFARE Classic part that broke. Detecting cards like this is ISO 14443.

Not sure how to go from here. I don't want to encourage you to break more tags, but it would be interesting to see if same thing happens if you use another phone with a card from the same badge. Also, to use your S7 with a card from another reseller/shop.

svedizz, 2019-01-19:
I don't have access to any other phones that this app supports so I haven't been able to do that yet. I ordered a new card a couple of days ago and see if the same thing happens again.

svedizz, 2019-01-19:
I just tried to write to a few sectors at a time and now it finally worked after disconnecting a few times. It seems that the tag gets disconnected when there's too much to read/write. It still doesn't make very much sense to me though.

ikarus23, 2019-02-01:
Great to hear you got it working, but yes, it is indeed strange. On the other hand, I've heard of strange issues like this quite often. There are way too much different Mifare Classic tags and Android "readers" out there. Also Magic Chinese tags are sometimes known for connectivity issues. I owe one with way worse antenna characteristics than normal tags. It has to be really close the reader.

svedizz, 2019-02-05:
My card just got bricked, and now it just tells me "No valid key found in chosen file."

svedizz, 2019-02-05:
https://www.imgur.com/a/fsdPgs5 This is what sector 0 looks like now, it also says that there are no keys anymore. When my normal public transit MiFare cards got bricked like this, they were still readable.

ikarus23, 2019-02-16:
> My card just got bricked

Sorry to hear that. 

> I thought these cards couldn't be bricked.

As far as I know there are tags that can be unbricked. The Proxmark3 has commands for this. But I never had these issues and I never tried the unbrick commands. Not sure if the tags can be unbricked from Android or if special commands are needed for this.

httpLov3craft, 2019-03-16:
I have some Gen 2 CUID cards that have the same problem. Fortunately the implementation is vulnerable to a nested attack. Ever few times I try to write them fully, or every time I try to write block 0 using MCT they loose connection. If it was a block 0 write I need only remove and approach the tag again. If it was a larger write, I need to use a nested attack through MFOC to find the seemingly random keys that were written blocking my sector access... once this is done I can use nfc-mfclassic to write a factory default card to the CUID card and recover it. Adding the random keys to MCT does not seem to allow me to recover the card in MCT.

I suspect this may be a combination of cheap crap cards, and error handling or how fast following APDU is sent in MCT. It makes it a little frustrating. I have some G2 UID cards from the same source that experience the same issue when writing through MCT (obviously without the block 0 writes), but at least those are easier to recover because of the backdoor.

ikarus23, 2019-03-16:
Loosing the connection while writing to block 0 is normal for some tags. Have a look at https://github.com/ikarus23/MifareClassicTool/issues/122.

It might be the cards that are causing this issue. But there is a chance it might be your phone. What device are you using?

httpLov3craft, 2019-05-03:
Was using a OnePlus 6T

ikarus23, 2019-05-03:
Do you have access to another device to test the same tags?

> have some G2 UID cards from the same source that experience the same issue when writing through MCT (obviously without the block 0 writes), but at least those are easier to recover because of the backdoor.

I never have used these type of tags (block 0 writable from Android AND backdoored). I'm not aware if there are known for causing trouble with MCT.

Kimo06, 2019-05-24:
Hello, I'm a newbie and it seems I have the same problem.
I'm using a Huawei p10 phone with usb ACR122U writer because my phone was not able to write block 0.
After many try I managed to write a mifare classic card which works, but I had many deconnections and I wrote it block by block.
I'm trying now with another dump. I have deconnections after block 0 and tried to write first or last.
When I compare block by block with my dump it is ok but doesn't work.
Do you think it is because I wrote block by block ?
Thanks for any help.

ikarus23, 2019-05-28:
Sorry to hear it does not work the way it's intended. This might be an issue of the External NFC app. However, there is nothing wrong with writing block by block. As long as the data gets written, you should get an identical copy. To write block 0 you need a special tag of course.

Kimo06, 2019-05-28:
Thank you for your response. I think I'm missing something, but when I compare sectors it seems to be correctly written and still does not work. I'll let you know if I find a solution.

Vaklin, 2019-05-28:
Just wrote an _one_time_block0_writable card. Bug, defined in tread 122 is still present. If I decide to write all sectors, the process starts with sector 0. Unfortunately, after change the content in block 0, the phone thinks here is a new tag. In my exact situation more than half of the memory wasn't wroten. In this case bricking the card is on one step distance. Enough is to write in some of blocks 3 defective access condition bytes and no more chance to use whole sector.

May I ask the owner of the software to share step by step activation procedure for chinese cards? I'd like to do something, but for now I have no success with halt command for few tags, bought as direct block 0 writable.

ikarus23, 2019-06-01:
>  Bug, defined in tread 122 is still present.

Yes, I don't know of any good way to fix this. Sorry.

> In my exact situation more than half of the memory wasn't wrote.

Didn't know this could happen. This is bad.

> May I ask the owner of the software to share step by step activation procedure for chinese cards?

I'm not sure what exactly it is you need. You should be able to clone a tag just by first writing block 0 only and then use the clone function to copy the rest. Or you get lucky buying another block 0 writable tag. I have a tag which works even if I do a full clone.

Vaklin, 2019-06-02:
Thank you for the response.
I'd like to propose you a way to solve this issue.

If the user wants to write in sector 0, the software checks is the same content in block 0 in the card and in the file to write. If it is the same, no problem at all. We will not write block 0 and so on. If it is different, we should check is the card accepts write operation to block 0. (BTW, this was my question, how to check is this card is a Magic card and how to activate it for write block 0.) Next step is to write only the block 0, inform the user about new UID and make a proposal to write the file again with the new parameters. In this situation block 0 will be the same and all be fine.

Kimo06, 2019-06-03:
I made 2 other keys :
- write tag/write dump(enable manufacturer writing)
-> error on sector 0 (only first line written)
- write tag again/write dump (same enable)
-> very fast, and compare to dump is ok, but key is not working
I am wondering does it mean that my dump is false ?? Is it possible, or is it because I need to write sector 0 twice ?
Thanks for any help.

Vaklin, 2019-06-03:
The problem should be in the answer from the tag to halt command. I have working tag and perfect copy from it. Original works, copied not. Dump is the same from both.

Please help me with the command sequence to PN 533 with register contents  to activate magic cards. There should be the weak place.

Vaklin, 2019-06-03:
About the block 0. More than normal for the phone is to found a new tag after change the UID. This is not a problem, just possible improvement in the app.

ikarus23, 2020-01-24:
> The problem should be in the answer from the tag to halt command. I have working tag and perfect copy from it. Original works, copied not. Dump is the same from both.
> 
> Please help me with the command sequence to PN 533 with register contents to activate magic cards. There should be the weak place.

What do you mean? The PN533 is supported by libnfc and there are tools like nfc-mfsetuid which will set the UID for backdoored Magic Chinese tags (1st gen).

ikarus23, 2020-01-24:
Also: This issue gotten a bit off topic. I will close it now. Feel free to still use it if you learned something new.

closed 2020-01-24 by ikarus23