ikarus23/MifareClassicTool #236: Does it compile for Android KitKat version?

akeilox, 2019-02-16:
I have Sony t3 which claims to read mifare classic cards. But runs on kit Kat. Can't update

If I compile this app (mifateclassic) will it produce KitKat compatible apk? Or to certain version only ?

ikarus23, 2019-02-18:
Can't you just install it? Anyways, even if you want to compile it your self, MCT works from Android 4.0 and up.

akeilox, 2019-02-19:
Hi @ikarus23 
Believe there was an issue with the sony t3 phone I received. I have since returned it and ordered a Sony Z3 D6603. After which I have seen the discussion thread on Z3 compatibility. Hopefully it'll work on Mifare classic as its listed under compatible devices.

Reason I have asked if it can compile on older version was I would like to make a simple app that can read MiFare tags, and have not ventured into this before. Is it an achievable/easy task, to incorporate reading MiFare tags, like in USB powered readers it reads the card on any active input textbox. 

If not I am hoping to learn on using the API and test cards with your application. Z3 is very affordable to get started for development it seems...

ikarus23, 2019-02-19:
Oh ok. Good luck with the Z3. And don't worry. Mifare Classic is not that hard. Read the [datasheet](http://www.nxp.com/documents/data_sheet/MF1S50YYX.pdf), some [Android NFC basics](https://developer.android.com/guide/topics/connectivity/nfc/) and use the [Mifare Classic API of Android](https://developer.android.com/reference/android/nfc/tech/MifareClassic).

closed 2019-03-16 by ikarus23