ikarus23/MifareClassicTool #240: Support for writing to block 0 of MIFARE Classic EV1 1K (7-byte UID)

raitoX, 2019-03-13:
This app is able to read from "MIFARE Classic EV1 1K" cards, so I attempted to make a copy by trying to write it to one of my magic cards (MIFARE Classic 1K).
I got the following error message:-
Error: BCC of block 0 is not valid

Is this control more of a software control or do I need the specific hardware? (do i need the specific MIFARE Classic EV1 1K magic card?)

This is the first line of block 0:-
04F9E4FAB35780884400C82000000000

ikarus23, 2019-03-16:
>Error: BCC of block 0 is not valid

This error only occurs if you try to write a UID with a wrong BCC to a 4 byte UID tag. If you write to a 7 byte UID tag the BCC is not checked. So what exactly are you doing? Are you sure you write to a 7 byte UID block 0 writable tag?

raitoX, 2019-03-17:
I see. I was trying to write a 7 Byte UID to a 4 byte tag.
I need to go find the correct tag then
Thanks!

ikarus23, 2019-03-21:
You are welcome!

closed 2019-03-21 by ikarus23