ikarus23/MifareClassicTool #243: Samsung S8 (SM-G950N)

jewangyi, 2019-04-11:
Using Android Pie, I can read and save tags fine. However, when trying to write the tags they do not work. Even formatting the tags do not work. I know it worked on Oreo but it doesn't work anymore. Any suggestions?

ikarus23, 2019-04-13:
Never heard of an issue like this. Until now it always was read and write works or read and write does not work. But only reading and not writing... strange...

Is there any error message? Does the tag use keys like `000000000000`?

jewangyu, 2019-04-14:
Just confirmed it using another phone of the same model. Same thing. It reads fine but doesn't write. It says it wrote it successfully but when I check it, it is still the same as before. I have tried several different tags, thinking it was the tags but they are all the same. Reads okay but doesn't write.

jewangyu, 2019-04-17:
So upon further testing, I can write to an NFC Card. However, I can't write to block 0 even though I have an UID changeable block 0 rewriteable card. I have tested several with the same results.

I was able to change the block 0 before but now I can't.

Any suggestions?

ikarus23, 2019-04-17:
Stupid question: You changed it before with your Android device, right? Not with a PC/laptop and RFID/NFC reader?

The-Clockmaker, 2019-04-18:
Samo here, all other blocks write fine, block 0 does not, even though it says it does.. Might be due to the cards though, I will test later on a different android device lower than 9.0

jewangyu, 2019-04-18:
Tested with a Samsung S7. Same situation, it doesn't write to block 0. 

Ikarus23:
I was able to write to block 0 using my phone before but now I can't. I am able to write to block 0 using an NFC reader/writer. So the cards and tags are fine. 

After the Samsung firmware update, I was no longer able to write to block 0.

Further, I have tested every version of MifareClassicTool from version 2.2.0 to 2.2.6. None of the versions can write to block 0.

It really blows since I used to use the app to clone many of my NFC cards. Now I have to carry several cards instead.

Any tip would be much appreciated.

ikarus23, 2019-04-18:
@The-Clockmaker: Yes, please test it with the other devices. In most cases it is the tag.

@jewangyu: How do you write it the tag using a NFC reader? Because there are several versions of UID changeable tags, but not all of them are compatible with Android.

jewangyu, 2019-04-19:
I have several different NFC cards (Mifare Ultralight, DESfire, etc). The tags and cards I am using with MifareClassicTool are Mifare Classic Block 0 writeable UID changeable.These cards are compatible with Android. As mentioned, I was able to use them with the app before to change block 0 to clone the cards but now I can't.

As for the reader/writer. It is a simple cloning device. I press a button and it read a card. I press another button and it clones the original card to the UID changeable card.

ikarus23, 2019-04-24:
Interesting, I thought the cloning devices do only work with the 1st gen UID changeable cards and not with the Android compatible 2ed gen block 0 writable cards. Would be best to find a well known and supported Android phone and try the same tag. It is strange enough that it does not work with a S8 and a S7 (although it worked in the past?!?).

The-Clockmaker, 2019-04-27:
Well it doesn't seem to work with any phones I've tried anymore so far, even severely outdated ones, so maybe it's the tags.. so monday I'm getting an acr122u in the mail and a few more tags from another vendor and I'll give those a go to see if it makes any difference

ikarus23, 2019-04-29:
I too think the tags are the issue here. Good luck trying the ACR122u. This reader is also able to change the UID of tags which are not Android compatible.

The-Clockmaker, 2019-04-30:
> 
> 
> I too think the tags are the issue here. Good luck trying the ACR122u. This reader is also able to change the UID of tags which are not Android compatible.

It turned out the tags are fine, using the acr122u and a script I was able to clone the tags and the uid,, block 0, all 64 blocks, but it doesn't work on any phone so far on MCT, I've tried a samsung s8, lg q6, lg g3, samsung a3, and samsung note 1 with the acr122u as external reader, none of them were a success..
However the tags I was trying to clone were hardened mifare classic tags, and I could only successfully clone them with a hardened nest attack to exploit sector 0, I don't know if that info makes any difference but k

ikarus23, 2019-05-03:
What script on kali? As far as I know the default is to clone using the "backdoor" 1st gen magic tag technique. In this case it can not work on Android. Only 2nd gen magic tags will work.

jewangyu, 2019-05-04:
Just tested new tags, still no good.

ikarus23, 2019-05-04:
@jewangyu Tested what? Writing to block 0? Or writing in general?

jewangyu, 2019-05-05:
I can't write to block 0. I can write to all other sectors.

ikarus23, 2019-05-05:
@jewangyu This must be an issue of the tag. It's either not a "magic tag 2nd gen" or the access conditions are bad. Or its is not a block 0 writable tag at all. MCT does nothing different when it comes to writing blocks. block 0 is treated the same as block 1, 2, ...

ikarus23, 2020-08-16:
I'm still convinced the tags are the issue here.

I will close this issue for now. Feel free to report back, if you learned something new about the situation.

closed 2020-08-16 by ikarus23

Undervalued, 2020-12-16:
Hi, i'm using a Xiaomi mi 10 lite 5g, and it doesn't work to clone UID, but in another phone (that has no nfc) with external nfc and ACR122u i can clone the UID in the same tag, so maybe it's not tag's problem

ikarus23, 2020-12-18:
This has nothing to do with this issue? Why write here?!? Please open a new issue.