ikarus23/MifareClassicTool #245: Old tag is disabled after use a new one

HeadFox, 2019-04-16:
I read and then write a dump to a new tag, after doing that, the new and the older one are working but only one time, just after the older one is disabled.
How is it possible ?
I tried to compare the dump and they are strictly identical...

HeadFox, 2019-04-17:
So I was reading a lot of forums and it appear that the sector 11 is written by the vigik reader. 
So the central know that the older badge don't have this new data....
Any workaround for this anti-copy feature ?

ikarus23, 2019-04-17:
Well it depends on what is written by the reader. I suppose it is some kind of timestamp or counter. But it might be encrypted with a secret you don't know. I don't think there is an easy way to make two tags work at the same time.

You can try to set the access bits of sector 11 to read-only. The reader should no longer be able to update sector 11 with the new data. However, I'm pretty sure a tag with this configuration is considered broken by the vigik system and will not work.

iceman1001, 2019-04-18:
I heared its a counter and its synced with backend.  Test to lock both tags sector11 but remember to write down old config first so you can reverse it back.

HeadFox, 2019-04-24:
Thanks ! But I'm afraid of one things. If I block write access but the central thinks that new block value is written and wait for it next time, I completely loose access... .

ikarus23, 2019-04-24:
You might be right. If you do not have the possibility to test it under "save" conditions you better not test it at all.

ikarus23, 2019-05-03:
I will close this issue now, but feel free to post any updates.

closed 2019-05-03 by ikarus23