ikarus23/MifareClassicTool #247: My device isn't supported, even with an external ACR122

joescorr, 2019-04-26:
My main issue here is the popup saying that my device isn't supported, even with an ACR122 plugged in. Any way I can disable that warning?

Thanks

ikarus23, 2019-04-29:
What device are you using? Have you installed External NFC?

joescorr, 2019-05-06:
Apologies for the delay,
I can get MCT to work with my USB reader, but the popup stating that my phone isn't supported (even with the USB plugged in and working) makes using MCT's UI a nightmare

ikarus23, 2019-05-06:
What exact error message is shown when?

Lexus89, 2019-08-16:
Any update about this? I am currently looking for a SCL3711 to connect to my Nexus 5 phone and hope MTC will work with that. From what i understand this should be possible right?

joescorr, 2019-08-16:
If your device isn't supported, it will show error messages saying that it
won't work, but will work just fine after you close them.
The error messages are more annoying than an actual issue

ikarus23, 2019-09-26:
Oh, ok. So you got External NFC installed and it works but when you start MCT it detects your device is not supporting Mifare Classic and show you a warning that you can not use it? Because MCT should check if External NFC is installed. What Android version are you using and what is the exact error message?

ikarus23, 2020-08-15:
I've just had a deeper look into the source code and I think I see what might cause this issue for some users. I will try to fix it and report back with a testing version for you guys to check out.

ikarus23, 2020-08-16:
I've build a testing version of MCT which should make this better. Please try it out! http://tests.icaria.de/MifareClassicTool-2.3.1-testing4.apk

ikarus23, 2020-09-01:
The fix is released with version 3.0.0. If it does not work, feel free to reopen this issue.

closed 2020-09-01 by ikarus23