ikarus23/MifareClassicTool #252: sak 28 is not compatible

zhengdianyong, 2019-05-06:
MCT can not read simulated card of Mi 9, because of the sak is 28, maybe.
Some information from NXP TagInfo:

# Detailed protocol information:
ID: xxxxxx
ATQA: 0x0400
SAK: 0x28
ATS: 0x05788070020000078069FFFFFFFFFFFF
* Max. accepted frame size: 256 bytes (FSCI: 8)
* Supported receive rates:
        - 106 kbit/s (DR: 1)
* Supported send rates:
        - 106 kbit/s (DS: 1)
* SFGT: 302.0 us  (SFGI: 0)
* FWT: 38.66 ms  (FWI: 7)
* NAD not supported
* CID supported
* Historical bytes: [none]

zhengdianyong, 2019-05-13:
May anyone help me?

ikarus23, 2019-05-13:
What is the error message?

ikarus23, 2019-05-13:
According to [AN10834](https://www.nxp.com/docs/en/application-note/AN10834.pdf) (page 7) a SAK of 0x28 is "Smart MX with MIFARE 1k". So your are correct, this is an emulated MIFARE Classic 1k. However, MCT uses the graph from the application note to determine the type. A SAK value of 0x28 should be parsed correctly and the tag should therefore be supported. Does your device not support MIFARE Classic?

zhengdianyong, 2019-05-16:
> According to [AN10834](https://www.nxp.com/docs/en/application-note/AN10834.pdf) (page 7) a SAK of 0x28 is "Smart MX with MIFARE 1k". So your are correct, this is an emulated MIFARE Classic 1k. [...]

When I put the emulated card by phone and another phone with MCT, nothing happened, even no errors found.
My device support MIFARE Classic, I can read the physical card, but cannot read the emulated card by a MI 9 Phone.

ikarus23, 2019-05-16:
Oh! So you emulate a MIFARE Classic card with a phone? I did not know there is a phone which is capable of this. Something must be wrong with the emulation. I don't think it is MCT.

zhengdianyong, 2019-05-17:
OK, I will try another way.
Thank you~

closed 2019-05-17 by zhengdianyong