ikarus23/MifareClassicTool #253: New phones that support NFC MIFare classic

maria1andersson, 2019-05-20:
Hello 
Thank you for this app 
PLZ what is New phones that support NFC MIFare classic ?

ikarus23, 2019-05-27:
Not sure about "new" phones. But you can have a look at https://github.com/ikarus23/MifareClassicTool/blob/master/COMPATIBLE_DEVICES.md

cosmocain, 2019-10-01:
Hello @ikarus23 , along with greeting and wishing that you continue contributing with the mct app, I have the doubt regarding a nokia 4.2 device will the app work with an android 9 for example?
What do you recommend to see the compatibility with the apk?
regards

ikarus23, 2019-10-03:
@cosmocain There is only one good way to check the compatibility. Get the device and an Mifare Classic tag. Then try to read and write to the tag. If it works everything should be fine.

Unfortunately there is no good way to tell if a device is supported without trying. There are only some list available, but they might be wrong:
* https://github.com/ikarus23/MifareClassicTool/blob/master/COMPATIBLE_DEVICES.md
* https://github.com/ikarus23/MifareClassicTool/blob/master/INCOMPATIBLE_DEVICES.md
* https://www.shopnfc.com/en/content/7-nfc-compatibility

cosmocain, 2019-10-03:
Hi @ikarus23 
Yesterday I did a direct test on a Nokia 6 device which I loaded the mct app and opened without any problem, also recognized the mifare label.
I give it to you as data for you to include in your list.
regards

alierta, 2019-12-27:
Hello I have a Huawei Mate 20 and mct works perfectly. Can read and write mifare classic without any issue with android 9. Tested by me. It can be added to compatible devices.

alierta, 2019-12-28:
And Oneplus 6T works too. I tested it. Thanks @ikarus23 for the app. If you need some screenshot or something I can provide it.

ikarus23, 2020-01-03:
Thanks! I will update the list of compatible devices.

comp500, 2020-01-04:
I have a Nokia 6.2 (Android 9), and it works as well. It seems likely that the 4.2 will support Mifare Classic if it has NFC, they probably use the same NFC chip.

ikarus23, 2020-01-24:
I guess this issue has been there for long enough. Don't forget to check the list of [compatible devices](https://github.com/ikarus23/MifareClassicTool/blob/master/COMPATIBLE_DEVICES.md).

closed 2020-01-24 by ikarus23