ikarus23/MifareClassicTool #255: Create Sectors in Dump Editor

ttepatti, 2019-06-24:
I have an issue with a card where every time I read it, different sectors fail. Sometimes it gets sector 5, for example, other times it fails it. Sometimes it might get 5 but fail on 12, or get 12 but fail on 3, etc.

I figured the best way to read this card would be to take five or six different dumps to ensure I get every sector, and then edit one of the dump files to add in any of the missing data, which I can find in the other dumps.

My only issue is: If a sector doesn't appear in a dump file, you can't add it in when you're editing the dump. If I take a dump and Sector 5 is missing, then when I go to edit that dump, I can't paste any new data into Sector 5 because it was missing when the dump was taken. It simply shows:

`` 
Sector: 4
[data here for sector 4]

Sector: 6
[data here for sector 6]
``

It would be really helpful if blank sectors were either shown as all zeros (or dashes or something) or if there was an "insert sector" button to insert data for a missing sector.

ikarus23, 2019-06-26:
I know this issue from my Nexus 5X. It is an issue within the NFC driver as far as I can say. There is not much I can do to fix it, but I've implemented a workaround. Use the "retry authentication" option from the settings menu. A value of 6 should work in most cases, but you can play around with it.

Regarding editing dumps: Just use a text editor. MCT dumps are just a text file with a very simple format. You can find an example  [here](https://github.com/ikarus23/MifareClassicTool/blob/master/tools/example-files/example-dump-file.txt). You can find the dumps on the external storage in the folder "MifareClassicTool".

ikarus23, 2019-09-26:
Closed. Lack of feedback.

closed 2019-09-26 by ikarus23