ikarus23/MifareClassicTool #256: Dump in App but not in Folder

H4xl0r, 2019-06-27:
Hey got a Strange situation and need help.
Got some dumps on my phone / the apps shows them and i am able to edit etc them.
but they are not shown under the app folder and dumps ?

where can i find those dumps ? need them !

Problem: 
Dumps are in the app , but not under appfolder (cant get the dumps out of the app) but need the dump file ....

ikarus23, 2019-06-29:
MCT dump files are by default on the external storage (e.g. the sdcard). For example `/sdcard/MifareClassicTool/dump-files`.

H4xl0r, 2019-06-29:
No SD vard inside the phone , thats the point searched the whole phone for those dumps , they are in the app but not in the storage .... Strange

ikarus23, 2019-07-03:
MCT uses the now depricated [getExternalStorageDirectory()](https://developer.android.com/reference/android/os/Environment.html#getExternalStorageDirectory()). There should be a specifed place where you find the files even with no sdcard presend.

HarvexARC, 2019-07-26:
Same questions happens here too.
Phone: Pixel XL Gen1 running google offical Andriod 9 without root
Using Google Play version 2.3.1

Old version of MCT and Android on my phone works fine before,
So I had "/sdcard/MifareClassicTool" folder, and some dump files, keys dictionarys already in it.
But now, in latest version, seems like MCT using another special location to save&read files.
Can't find my old files in MCT now, and any new files created by MCT is not save to "/sdcard/MifareClassicTool" as before.
And the storage space using  by MCT in Andriod 9 is invisable to none rooted users, and USB MTP transfer.
So there is no way to do any files export&import by now for non rooted users. 
Hope this can be fix because this problems really make MCT unusable.

HarvexARC, 2019-07-26:
Sorry!
Just found that normal free version in google play dosn't have this problem.
This is just happend in donate version.
That's not a problems any more

ikarus23, 2019-07-28:
What?!? The donate version is exactly the same as the normal version. This is really strange.  @H4xl0r do you also use the donate version and also Android 9?

H4xl0r, 2019-07-28:
no android 6 and not the donation version , still unable to find the dump , strange

HarvexARC, 2019-07-29:
Oh sorry, I had misunderstanding on Use Internal Storage setting.And I can read&write my previous files by uncheck this setting. But Pixel XL still got freeze. Now I'm working with my new Xperia 1, Everything is perfectly fine.

ikarus23, 2019-07-30:
>Oh sorry, I had misunderstanding on Use Internal Storage setting.

Hehe, yes, if this setting is activated the internal app storage is used. This storage can not be accessed by any other app. Therefore you can not find the files with an file browser app.

ikarus23, 2019-09-01:
Hi @H4xl0r, any news? Have you checked your MCT options? Have you tried to connect your phone to a PC and look for the data?

ikarus23, 2020-01-24:
Closed. Lack of feedback.

closed 2020-01-24 by ikarus23