ikarus23/MifareClassicTool #260: Limited number of dumps

michto36, 2019-08-05:
Hi, 
This is my issue:

i can't put more than 145 files into "dump-files" folder, and no more 140 files into "key-files" folder on my device. is it normal ? is there a way to extend this limit ?

Thanks

ikarus23, 2019-08-06:
Hi, this is interesting. I never thought someone would have so may dump/key files. However, I never limited the number by intention, so I'm not sure what is causing this issue. When I have some time I will look into it.

michto36, 2019-08-22:
Ok, no worries, thank you for spending time on this issue ;-)

ikarus23, 2019-08-24:
Hi, how exactly do you notice the limitation? Does clicking on the "Edit Dump" button no longer work? Do you want to save another tag and there is an error? Because I'm able to have 1500 files in my dump directory without any issues...

michto36, 2019-08-24:
Hi !
I noticed this limitation when i transfert the dumps into phone internal memory with the « AndFTP » application. It connects to my ssh server and gets all dumps i have in my folder. And after launching MCT application, i can see limited number of dumps.

Maybe should i have to change the way of transfering dumps into the phone. Which way are you using ?

michto36, 2019-08-25:
I tried to sync my dumps directly to the phone with windows explorer, and same result. It copies the files but at the end only 145 files are in my « dump-files » folder..

If it can help, I am on Samsung Galaxy S5 , Android Marshmallow 6.0.1, and MCT version 2.3.1.

ikarus23, 2019-08-26:
Hmm, I'm starting to believe this is an issue related to your Phone/OS and not MCT. Have you tried to create another folder on your external storage and copy the files there?

michto36, 2019-09-04:
Hi !
I think you were right about my phone ! I have resetted the whole phone and reinstalled MCT at last version and now it works like a charm ! 
Sorry for the inconvenience, and thank you for your time ! 

Good work with this application.

closed 2019-09-04 by michto36

ikarus23, 2019-09-04:
Thanks for the feedback and your kind words. Have great day.