ikarus23/MifareClassicTool #261: Can you allow Galaxy S7 Active?

padillla, 2019-08-14:
I have a galaxy S7 active, the other S7 are listed in the compatible, but for some reason my s7 whows as incompatible, please modify the Play settings to allow Samsung SM-G891A to use this app. 

Thank you

ikarus23, 2019-08-15:
Well, have you tested MCT on the S7 Active? Did it start to work since Android 8?

padillla, 2019-08-19:
I'd like to

ikarus23, 2019-08-19:
I thought you already did ;) Just install the app from http://publications.icaria.de/mct/releases/.

ikarus23, 2019-08-19:
And does the S7 Active have Android 8?

padillla, 2019-08-20:
Yes, it has android 8, I will download it from there.

This app, allows cloning or reprogram of HID cards, right?

thank you

ikarus23, 2019-08-23:
No. It is for Mifare Classic cards only. That's why it's named "Mifare Classic Tool".

ikarus23, 2019-08-27:
I guess this done now.

closed 2019-08-27 by ikarus23