ikarus23/MifareClassicTool #264: Folders for dumps

rawisha, 2019-08-29:
Hello Rawi here.

Not sure if this is an issue than more of a suggestion.

I was thinking of making a folder for different dumps i have on the app, would it be possible to accomplish that in the near future?

I did try to make a folder and open it on the app, but it reads it as a dump sadly.

i have alot of dumps and i would like to organize it somehow if possible.

Thanks, Regards

Rawi.

rawisha, 2019-08-29:
Oh and also I have few questions about  dump I have.
Is there a way I can contact you?

iceman1001, 2019-08-30:
@ikarus23   .. the rrg/iceman repo has some default folder to save/load stuff from now.
I just wanna make sure that on android both MCT / rrg app  is aware about each different places.
so we can search there also.   What is your take on that idea?

rawisha, 2019-08-30:
I was thinking of having private folders

ikarus23, 2019-09-01:
Hi!
Interesting feature request. I never thought people would going to use this app so much that they need folders for organizing. :) However, my file chooser is really simple and the rest of the app (e.g. the saving function) is also not aware of folders. Making MCT "folder aware" is not that easy. Well... ok it is easy, but it is also time consuming and I'm not willing to put the time in just now. I have a lot of other projects ;) Sorry.

Maybe the best way will be to use a file management library. So far there are no external dependencies in MCT, but this might be the point were it is the best thing to do. Because once you have folder-based dump management, you want the same for keys. And you want to move them, copy them, etc. There should be good libraries for this.

Regarding "private" folders: MCT by default saves everything to your external storage in order to make it easy for you to import/export files. There is an option which lets you choose to use the internal (private) storage. However, you will not be able to import files to there. Exporting is only possible using the "share" option. This is for now. I'm working on better import/export function which will also be able to convert file types. Not sure when this will be ready.

@iceman1001 Are the folders of the rrg app on the external storage? Because if the app saves its data to its app specific storage (internal storage), there is no way users can backup files them or import them to MCT.

iceman1001, 2019-09-01:
As I understand it  the RRG app saves to SD card.