ikarus23/MifareClassicTool #267: Nothing happens when mapping on Galaxy S7 SM-G930U latest firmware

Gigelf, 2019-09-25:
Hi,

I bought a Galaxy S7 to use MCT.
I got a SM-G930U (usa firmware), patched it with latest june patch.
When I start "mapping and read tag", nothing happens, continuous wheel. 
NXP TagInfo "freezes" same way but NFC Tools can read without problem.
A friend of mine as a European version of the S7 (SM-G930F), and it works perfectly.

Any idea from what it can be? NFC chip can read mifare because NFC tools app can.

Thanks

Gigelf, 2019-09-25:
Here are some debug logs, if it can help
[logcat.txt](https://github.com/ikarus23/MifareClassicTool/files/3653825/logcat.txt)
[run.txt](https://github.com/ikarus23/MifareClassicTool/files/3653826/run.txt)

ikarus23, 2019-09-26:
Very interesting. At this point I'm have no idea what could cause this. It is weird that an official app from the manufacturer of Mifare Classic tags can not read them, but a third party app can.

Does the tag have default keys? Can NFC Tools display the content of the whole tag?

Gigelf, 2019-09-28:
Tried a blank card with default keys and a door tag with keys i cracked.
Can't test door tag to read whole content on NFC Tools as it has a non default key.

MCT Video: https://www.youtube.com/watch?v=y8dc3bAaq9E&feature=youtu.be

NFC Tools video: https://www.youtube.com/watch?v=T3Vv0TpX18s&feature=youtu.be

Gigelf, 2019-09-29:
Forgot to point out something: the nfc service seems to crash when trying to read a tag. I have to uncheck/check the nfc service to be able to read an other tag. This problem does not appear with NFC Tools.

ikarus23, 2019-10-03:
I can't watch the videos, they are marked private.

Gigelf, 2019-10-03:
Oops, repaired that.

masongaffney, 2020-03-20:
I can’t even get mine to flash now and I think it’s bricked I might of used the wrong firmware. But I have the same phone as you.

ikarus23, 2020-08-15:
Sorry for not reporting back earlier. It looks like the tag has `000000000000` as key. This causes issues on some devices. Maybe your model/firmware of the S7 has this issue. There is a thread over at https://github.com/ikarus23/MifareClassicTool/issues/66. Especially https://github.com/ikarus23/MifareClassicTool/issues/66#issuecomment-478218028 is worth a read. But I'm still puzzled why NFCTools can read something...

Fbrachere, 2020-09-13:
Hi, I've exactly the same problem here with my S7 SM-G930U.
I can't read any tag memory but I can write a block on it with my custom keys.

ikarus23, 2020-09-13:
> I can't read any tag memory
Have you tried reading new/empty/factory formatted tag?

Fbrachere, 2020-09-13:
> Have you tried reading new/empty/factory formatted tag?

No, I didn't. I just tried and it works for an empty tag.

ikarus23, 2020-09-13:
Hmm. Again, this sounds like the key `000000000000` issue from https://github.com/ikarus23/MifareClassicTool/issues/66. I don't think there is much I can do. At least not without owning a Galaxy S7 for debugging.

Fbrachere, 2020-09-13:
> I don't think there is much I can do. At least not without owning a Galaxy S7 for debugging.

This is my only phone, I can't send it to you but I can do some tests if you want. I'm able to use adb and compile Android apps if necessary.

ikarus23, 2020-09-24:
I guess the first step is to verify, if the tag in question has a key of `000000000000`. Do you have e.g. a USB reader to try it?

Fbrachere, 2020-09-25:
> I guess the first step is to verify, if the tag in question has a key of `000000000000`. Do you have e.g. a USB reader to try it?

Yes, I've one.
The tag does not have a `000000000000` key. Only `FFFFFFFFFFFF`, `EDEDEDEDEDED` and a random one.

ikarus23, 2020-09-26:
Strange. So a factory formatted tag can be read, but this one not, even though it has "normal" keys. If you know your way around Andorid Studio, Java and debuggers you can try to set a break-point somewhere around [here](https://github.com/ikarus23/MifareClassicTool/blob/master/Mifare%20Classic%20Tool/app/src/main/java/de/syss/MifareClassicTool/MCReader.java#L542). At least this is the section where the authentication is for the key mapping process is. But I'm not sure what to expect. I've never head this error and I have no clue what is causing it.

Fbrachere, 2020-09-29:
Thanks for your advices, I found the root cause:
The call to transceive from the function authenticate in [MifareClassic.java](https://developer.android.com/reference/android/nfc/tech/MifareClassic#transceive(byte[])) never returns when you use a wrong key.
I tried to hard-code the mKeyMap with the keys of my tag, short-circuit [buildNextKeyMapPart](https://github.com/ikarus23/MifareClassicTool/blob/master/Mifare%20Classic%20Tool/app/src/main/java/de/syss/MifareClassicTool/MCReader.java#L515) and it works, I can read the tag.
When I use wrong keys, I've got a lot of error in logcat even when MFCT is terminated:
```2020-09-29 07:01:45.042 7994-8070/? E/BrcmNfcJni: getReconnectState = 0x0
2020-09-29 07:01:45.053 7994-8016/? E/BrcmNfcJni: setReconnectState = 0x0
2020-09-29 07:01:45.058 7994-8016/? E/NxpExtns: Mifare Error in payload response
2020-09-29 07:01:45.058 7994-8016/? E/BrcmNfcJni: setReconnectState = 0x0
2020-09-29 07:01:45.065 7994-8070/? E/BrcmNfcJni: getReconnectState = 0x0
2020-09-29 07:01:45.072 7994-8016/? E/BrcmNfcJni: setReconnectState = 0x0
2020-09-29 07:01:45.081 7994-8016/? E/NxpExtns: Mifare Error in payload response
2020-09-29 07:01:45.082 7994-8016/? E/BrcmNfcJni: setReconnectState = 0x0
2020-09-29 07:01:45.087 7994-8070/? E/BrcmNfcJni: getReconnectState = 0x0
2020-09-29 07:01:45.098 7994-8016/? E/BrcmNfcJni: setReconnectState = 0x0
2020-09-29 07:01:45.103 7994-8016/? E/NxpExtns: Mifare Error in payload `response
```
I must deactivate and reactivate NFC in Android to stop errors.

I don't know why transceive function doesn't return with the wrong key, maybe a bug or a "security" feature.
So if I want to use MifareClassicTool, I need to know the whole key mapping, and to know the key mapping I need to read the tag ....  this is a vicious circle.
The two solutions I see:
- Add a feature to MifareClassicTool to use a complete key mapping file (keys A and B from all sectors), useless for many users I think.
- Find why transceive function don't work with wrong key.

ikarus23, 2020-09-29:
Thanks for investigating! For me this looks like a bug in the NFC stack. Not the first time something like this comes up. Regarding your solutions:
1. This is already on the roadmap for quite some time... https://github.com/ikarus23/MifareClassicTool/issues/43
2. Even if you find the bug, Android or Samsung has to fix it, I guess.

ikarus23, 2020-10-16:
After some reading: [transceive(byte[])](https://developer.android.com/reference/android/nfc/tech/MifareClassic#transceive(byte[])) can block until the tag is done with the operation and answers. So there might be a chance that the tag is broken (only answers if key is correct). The easiest way to check this would be to use a Proxmark3 and sniff. However, I don't think this is very likely.

There is a [setTimeout(int)](https://developer.android.com/reference/android/nfc/tech/MifareClassic#setTimeout(int)) function to cancel the `transceive()`. It may be used to improve the user experience, but it won't fix the issue.

I don't this there is anything helpful I can do for now. Therefore I'm closing this issue. Feel free to reopen it, if you find any promising lead on how to fix it.

closed 2020-10-16 by ikarus23