ikarus23/MifareClassicTool #268: unknown tag

NaharD, 2019-09-26:
I writed wrong data for block0 and my tag became not writable, but it works and opens the door.
Not enough setting for forced writing without check tag type.
To be able to correct mistakes made

ikarus23, 2019-09-26:
What do you mean by "check tag type"? What checks are prevent you from writing block 0 again? Did you change the access bits as well?

NaharD, 2019-09-26:
Yes, and my tag has become unknown type. I get an error while writing - there is no mifare classic tag.

I mean, checkbox in settings, which permited write on no mifare classic tag, like GOD mode.

Or validation SAK and ATQO on write block0, this work for BCC

ikarus23, 2019-09-28:
You most likely wrote faulty data to the tag. What exactly did you write? Could you provide the dump? And what device did you use?

NaharD, 2019-09-28:
Yes, you right. I wrote wrong data in section1 and block1
`B5F4E4E420000000000000000000000`
checksum 42 will be calculated internal  BCC tool, but other data will be set as zero, because I didn't fully understand why she needed it.

ikarus23, 2019-09-28:
Yes, the BCC is wrong. It should be `41`. Most Android phones are not able to read a tag with a wrong BCC value. But you can fix it with an external reader like the ACR122u. It ignores the BCC and can write a correct block 0.

NaharD, 2019-09-28:
Oh, tool generate wrong number for me? https://prnt.sc/pc5tet

ikarus23, 2019-09-28:
`B5F4E4E4` is not the same as `B5F74E4E`.

NaharD, 2019-09-28:
sorry, i wrong. Right - B5F74E4E

ikarus23, 2019-09-29:
So you have used `B5F74E4E` with `42` as BCC and the rest is all `00`? An now the tag does no longer work?

NaharD, 2019-09-29:
No, it is work, but tag gan unknown type and him not writable and not readable. Error like -  there is no mifare classic tag.

https://prnt.sc/pciksg
https://prnt.sc/pcikxp

ikarus23, 2019-10-03:
Ok, so it looks like your SAK value is wrong. Some tags allow to change the SAK value. Maybe one of the `00` you put after the UID+BCC is representing the SAK. But I'm not sure about that. However, there is nothing I can do. There is not way in Android to tell the system to ignore the SAK. Maybe you can fix your card using a PC and a RFID reader.

NaharD, 2019-10-03:
I have RC522 reader, but his not writed this tag.

Thought your application has a tag type check, but it OS level and not application

Thanx, for help)

ikarus23, 2019-10-03:
You are welcome. The RC522 should be able to read and write Mifare Classic tags.

closed 2019-11-03 by NaharD