ikarus23/MifareClassicTool #271: dump editor on the Meizu m16S

Belllock, 2019-09-27:
encountered such a bug:
I tried to change the screen scale and font size - it does not help. what can be done so that 32 characters are in one line!?
Thank!
[image]

[image]

ikarus23, 2019-09-28:
This is strange. Does [this testing version](http://tests.icaria.de/MifareClassicTool-2.3.1-testing1.apk) work better?

ikarus23, 2019-09-28:
Or maybe [this testing version](http://tests.icaria.de/MifareClassicTool-2.3.1-testing2.apk)? On the first the text will most likely not be centered.

Belllock, 2019-09-28:
No. Nothing changed. The second version only centers the text

[image]

ikarus23, 2019-10-03:
Hmm, this was somehow expected. But it should work in the first place. It might be a bug in the Android ROM of your phone. The text boxes for each sector have the layout attribute `WRAP_CONTENT` so the should be as wide as the text within it. Never heard of anybody with the same issue... Do you have the possibility to install another ROM to your phone? Just to check...

ikarus23, 2019-10-03:
Although I think your ROM is broken, I tried another workaround. Please test it: http://tests.icaria.de/MifareClassicTool-2.3.1-testing3.apk

Belllock, 2019-10-04:
nothing changed. :( I'll try to ask other owners Meise 16s about this problem). Thanks for the help!

[image]

ikarus23, 2019-10-26:
In that case I don't think there is much I can do. It seams something is wrong with Android on the device. It just won't wrap the content of text input fields correctly.

ikarus23, 2020-01-24:
I'm closing this because there is nothing I can do. Sorry. Feel free to provide feedback if you manged to fix it. Maybe an future Android update will make this right.

closed 2020-01-24 by ikarus23