ikarus23/MifareClassicTool #277: Error: BCC of block 0 is not valid

Bulte01, 2019-11-29:
Hello Everyone,

When I try to write to my card I get the error "Error:BCC of block 0 is not valid".
I have this card to write to [https://www.ebay.nl/itm/5PCS-UID-Changeable-Keyfob-Compatible-with-MCT-Block-0-Direct-Writable-by-Phone/272654995341?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649](url)

Kind regards,

Bas

ikarus23, 2019-12-06:
The tags look good. But data you are trying to write is not correct. Like the error says, the BCC is not correct. For MIFARE Classic tags with a 4 byte UID, the BCC must be the 5th byte of the very first block (manufacturers block). This app has BCC calculator tool. More information about calculating the BCC and how it is used during the anti-collision phase can be found in NXP's [AN10927](https://www.nxp.com/docs/en/application-note/AN10927.pdf).

ikarus23, 2020-01-24:
Any news on this?

Bulte01, 2020-01-25:
> Any news on this?

Nope cannot find the solution

ikarus23, 2020-01-25:
I did wrote what is causing this error message, so you should be able to fix your data.

Bulte01, 2020-01-25:
> I did wrote what is causing this error message, so you should be able to fix your data.

I have no idea how to fix it and I have not many experience with NFC chips

ikarus23, 2020-01-26:
Like I said, you have to make sure the data you are writing is correct. In your case, block 0 is not correct. If you just want to copy the UID, the easiest way it to use the "Clone UID" tool.

ikarus23, 2020-08-15:
I will close this for now. You should be able to get everything going. Just take your time and learn more about MIFARE Classic. ;)

closed 2020-08-15 by ikarus23